RemixVSL/iomemory-vsl4 #1: Error building package on Ubuntu 19.10

currowth, 2020-04-05:
Hi Funs,
Thank you so much for updating vsl4 for 5.x kernels.
I had to use the below build commands to build DKMS. I was able to see the disk with fdisk on ubuntu 19.10 with 5.3 kernel.
#---DKS Command-----#
sudo cp -r iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 /usr/src/
sudo mkdir -p /var/lib/dkms/iomemory-vsl4/4.3.7/build
sudo ln -s /usr/src/iomemory-vsl4-4.3.7 /var/lib/dkms/iomemory-vsl4/4.3.7/source
sudo dkms build -m iomemory-vsl4 -v 4.3.7 
sudo dkms install -m iomemory-vsl4 -v 4.3.7 
sudo modprobe iomemory-vsl4

When I ran dpkg-buildpackage, i got the below error. Is it possible to use the util without building the package?

dh_installdirs: This feature will be removed in compat 12.
# Add here commands to install the arch part of the package into
# debian/tmp.
dh_install --sourcedir=root --fail-missing -X/usr/share/doc/iomemory-vsl4
dh_install: Compatibility levels before 9 are deprecated (level 5 in use)
dh_install: Please use dh_missing --list-missing/--fail-missing instead
dh_install: This feature will be removed in compat 12.
dh_install: Cannot find (any matches for) "usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.mod.c" (tried in root)

dh_install: iomemory-vsl4-source missing files: usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.mod.c
dh_install: Cannot find (any matches for) "usr/src/iomemory-vsl4-4.3.7/kfio/.x86_64_cc48_libkfio.o.cmd" (tried in root)

dh_install: iomemory-vsl4-source missing files: usr/src/iomemory-vsl4-4.3.7/kfio/.x86_64_cc48_libkfio.o.cmd
dh_install: missing files, aborting
make: *** [debian/rules:139: install-arch] Error 255
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2

snuf, 2020-04-06:
Hi @currowth 

Thanks for reporting this bug!

Yeah the module works without the dpkg stuff if that's what you're asking. The dpkg build seems somewhat borked. Even when removing that file from the debian package build files it still barfs, which leaves me to believe that the whole thing is just broken. After removing the missing file it bombs out on:
```
dh_install: Cannot find (any matches for) "usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.mod.c" (tried in root)

dh_install: iomemory-vsl4-source missing files: usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.mod.
```
Think some time needs to be spent on overhauling the original broken debian packaging, the iomemory-vsl builds but doesn't include the module. So fixing both would be ideal. I'll have a look at it.

Best,

Funs

Tourman36, 2020-04-10:
I can get dpkg-buildpackage to work under proxmox-ve which uses a 5.3 Kernel. Are you running it from the root folder? If you run it anywhere else it doesn't work.

snuf, 2020-04-10:
Should be fixed now, the source.install file was the problem next to a couple of other things in the package build process that were off (bumped to version 9 instead of 5). Tested with ```dpkg-buildpackage -rfakeroot -b -uc``` from the checked out root of the project:

```
snuf@scipio:~/tmp$ git clone git@github.com:snuf/iomemory-vsl4.git
snuf@scipio:~/tmp$ cd iomemory-vsl4/
snuf@scipio:~/tmp/iomemory-vsl4$ git checkout backport-iomemory-vsl 
snuf@scipio:~/tmp/iomemory-vsl4$ dpkg-buildpackage -rfakeroot -b -uc
snuf@scipio:~/tmp/iomemory-vsl4$ ls -1la ../iomemory-vsl4*
-rw-r--r-- 1 snuf snuf    7621 Apr 10 10:36 ../iomemory-vsl4_4.3.7.1205-1.0_amd64.buildinfo
-rw-r--r-- 1 snuf snuf    3093 Apr 10 10:36 ../iomemory-vsl4_4.3.7.1205-1.0_amd64.changes
-rw-r--r-- 1 snuf snuf  837508 Apr 10 10:36 ../iomemory-vsl4-5.3.0-42-generic_4.3.7.1205-1.0_amd64.deb
-rw-r--r-- 1 snuf snuf   13356 Apr 10 10:36 ../iomemory-vsl4-config-5.3.0-42-generic_4.3.7.1205-1.0_amd64.deb
-rw-r--r-- 1 snuf snuf  531568 Apr 10 10:36 ../iomemory-vsl4-di-5.3.0-42-generic_4.3.7.1205-1.0_amd64.udeb
-rw-r--r-- 1 snuf snuf   11224 Apr 10 10:36 ../iomemory-vsl4-initrd-5.3.0-42-generic_4.3.7.1205-1.0_amd64.deb
-rw-r--r-- 1 snuf snuf 2574912 Apr 10 10:36 ../iomemory-vsl4-source_4.3.7.1205-1.0_amd64.deb
```
Let me know if it works now, and we can close the ticket.

Tourman36, 2020-04-10:
works fine now under Kernel 5.3 in proxmox-ve

`make[1]: Leaving directory '/home/src/iomemory-vsl4'
 dpkg-genbuildinfo
 dpkg-genchanges  >../iomemory-vsl4_4.3.7.1205-1.0_amd64.changes
dpkg-genchanges: info: including full source code in upload
 dpkg-source --after-build .
dpkg-buildpackage: info: full upload; Debian-native package (full source is included)`

currowth, 2020-04-11:
I was able to successfully build the below packages on on ubuntu 19.10.
After installing built packages, and the fio packages from sandisk, the fio commands shows the device properly.

When i try to install iomemory-vsl4-source_4.3.7.1205-1.0_amd64.deb, i get the below error. 
```
dpkg: regarding iomemory-vsl4-source_4.3.7.1205-1.0_amd64.deb containing iomemory-vsl4-source:
 iomemory-vsl4-source conflicts with iomemory-vsl4
  iomemory-vsl4-5.3.0-46-generic provides iomemory-vsl4 and is present and installed.
```

The other packages installed fine. iomemory-vsl4-source_4.3.7.1205-1.0_amd64.deb is not required right?

snuf, 2020-04-12:
@currowth nah the source package is only required if you want have a distributable source package. 

Closing this one, thanks for the testing and feedback!

closed 2020-04-12 by snuf