RemixVSL/iomemory-vsl4 #2: DKMS build error

naamval, 2020-04-08:
Hi Funs, thanks for your efforts to make this expensive paperweight actually useful! It seems that I'm hitting a wall when trying to do the DKMS build, unfortunately. After doing the `dkms build -m iomemory-vsl4 -v 4.3.7`, I get:
```
'make' DKMS_KERNEL_VERSION=5.3.18-3-pve.............(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.3.18-3-pve (x86_64)
```

The logs show:

```
[...]

make[1]: Entering directory '/usr/src/linux-headers-5.3.18-3-pve'
printf '#include "linux/module.h"\nMODULE_LICENSE("Proprietary");\n' >/var/lib/dkms/iomemory-vsl4/4.3.7/build/license.c
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/main.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/dbgset.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/license.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/pci.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/sysrq.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/driver_init.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kfio.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/errno.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/state.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kmsg.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/six_lock.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kcache.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kfile.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kmem.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kscatter.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/ktime.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/sched.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/cdev.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kscsi.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kscsi_host.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kcpu.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kcondvar.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/common_kinfo.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kinfo.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kinit.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kfio_common.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/khotplug.o
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kcpu.c:246:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
     ^~~~~~~~
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kcpu.c:246:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/kcsr.o
  SHIPPED /var/lib/dkms/iomemory-vsl4/4.3.7/build/kfio/x86_64_cc83_libkfio.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/module_param.o
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c: In function ‘fio_queue_rq’:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:625:23: warning: unused variable ‘disk’ [-Wunused-variable]
     struct kfio_disk *disk = hctx->driver_data;
                       ^~~~
In file included from /var/lib/dkms/iomemory-vsl4/4.3.7/build/include/fio/port/kfio.h:43,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/port-internal-boss.h:68,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/port-internal.h:57,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:66:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c: In function ‘kfio_dump_bio’:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:1544:45: warning: passing argument 1 of ‘bio_segments’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
              msg, BI_IDX(bio), bio_segments(bio), BI_SIZE(bio));
                                             ^~~
/var/lib/dkms/iomemory-vsl4/4.3.7/build/include/fio/port/common-linux/kfio.h:69:63: note: in definition of macro ‘kfio_kprint’
      kfio_print("%s" prefix fmt, kfio_print_prefix[level], ## __VA_ARGS__)
                                                               ^~~~~~~~~~~
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:1543:5: note: in expansion of macro ‘infprint’
     infprint("%s : idx: %x : phys_segments: %x : size: %x",
     ^~~~~~~~
In file included from ./include/linux/blkdev.h:21,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/port-internal.h:59,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:66:
./include/linux/bio.h:174:49: note: expected ‘struct bio *’ but argument is of type ‘const struct bio * const’
 static inline unsigned bio_segments(struct bio *bio)
                                     ~~~~~~~~~~~~^~~
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c: In function ‘kfio_alloc_queue’:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:2368:16: warning: passing argument 1 of ‘memcpy’ discards ‘volatile’ qualifier from pointer target type [-Wdiscarded-qualifiers]
         memcpy(&dp->queue_lock, &rq->queue_lock, sizeof(dp->queue_lock));
                ^~~~~~~~~~~~~~~
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/slab.h:15,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/include/fio/port/linux/kfio_config.h:176,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/include/fio/port/kfio_config.h:42,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/port-internal-boss.h:62,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/port-internal.h:57,
                 from /var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:66:
./include/linux/string.h:347:37: note: expected ‘void *’ but argument is of type ‘volatile fusion_spinlock_t *’ {aka ‘volatile struct __fusion_spinlock *’}
 __FORTIFY_INLINE void *memcpy(void *p, const void *q, __kernel_size_t size)
                               ~~~~~~^
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c: At top level:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:617:20: warning: ‘kfio_request_to_bio’ declared ‘static’ but never defined [-Wunused-function]
 static kfio_bio_t *kfio_request_to_bio(kfio_disk_t *disk, struct request *req,
                    ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:484:13: warning: ‘kfio_restart_queue’ defined but not used [-Wunused-function]
 static void kfio_restart_queue(struct request_queue *q)
             ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c: In function ‘fio_queue_rq’:
/var/lib/dkms/iomemory-vsl4/4.3.7/build/kblock.c:635:8: warning: ‘fbio’ is used uninitialized in this function [-Wuninitialized]
     if (!fbio)
        ^
  LD [M]  /var/lib/dkms/iomemory-vsl4/4.3.7/build/iomemory-vsl4.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: could not find /var/lib/dkms/iomemory-vsl4/4.3.7/build/kfio/.x86_64_cc83_libkfio.o.cmd for /var/lib/dkms/iomemory-vsl4/4.3.7/build/kfio/x86_64_cc83_libkfio.o
FATAL: modpost: GPL-incompatible module iomemory-vsl4.ko uses GPL-only symbol 'ktime_get_real_seconds'
make[2]: *** [scripts/Makefile.modpost:92: __modpost] Error 1
make[1]: *** [Makefile:1659: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.18-3-pve'
make: *** [Makefile:116: modules] Error 2
```

Config: proxmox kernel 5.3.18-3-pve, SX350-1300.

snuf, 2020-04-08:
@naamval ah yes the DKMS build didn't build GPL by default, corrected it, so it should build now.

naamval, 2020-04-09:
It works now! Was able to build the iomemory-vsl4 module, attach the fio card and mount the old ZFS filesystem that's stored on it. Thanks! If there's anything else I could test, let me know.

snuf, 2020-04-09:
cool, appreciate the testing and effort! If there are any tests you can think of that might be useful? I have the ones that are done when testing here: https://github.com/snuf/hakmotif/tree/master/module_tester/scripts with fio, for integrity and base functionality, just to make sure nothing blows up.

closed 2020-04-10 by snuf