RemixVSL/iomemory-vsl4 #11: rpmbuild fails in Fedora 31

Tourman36, 2020-04-21:
```
[root@localhost iomemory-vsl4]# rpmbuild -ba fio-driver.spec
warning: line 37: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
warning: line 37: It's not recommended to have unversioned Obsoletes: Obsoletes: iodrive-driver-kmod, iodrive-driver, fio-driver
error: Bad source: /root/rpmbuild/SOURCES/iomemory-vsl4-4.3.7.1205.tar.gz: No such file or directory

```

Tourman36, 2020-04-21:
> [root@localhost iomemory-vsl4]# git checkout code-cleanup
> Branch 'code-cleanup' set up to track remote branch 'code-cleanup' from 'origin'.
> Switched to a new branch 'code-cleanup'
> [root@localhost iomemory-vsl4]# make dkms
> cd root/usr/src/iomemory-vsl4-4.3.7 && \
>         make dkms
> make[1]: Entering directory '/home/temp/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
> ./module_operations.sh -n "iomemory-vsl4" -v "36bab4a" -d
> Adding module version v5.6.0-4.3.7 to source license.c
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc41_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc43_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc44_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc46_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc48_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc49_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc54_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc63_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Ok: patched version for /usr/src/iomemory-vsl4-v5.6.0/kfio/x86_64_cc75_libkfio.o_shipped replaced 4.3.7.1205 bryce@a7e621e6fb50 with v5.6.0-4.3.7.1205
> Adding, buidling and installing iomemory-vsl4/v5.6.0 with DKMS
> 
> Creating symlink /var/lib/dkms/iomemory-vsl4/v5.6.0/source ->
>                  /usr/src/iomemory-vsl4-v5.6.0
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> 'make' gpl DKMS_KERNEL_VERSION=5.5.17-200.fc31.x86_64......
> cleaning build area...
> 
> DKMS: build completed.
> 
> iomemory-vsl4.ko.xz:
> Running module version sanity check.
>  - Original module
>    - No original module exists within this kernel
>  - Installation
>    - Installing to /lib/modules/5.5.17-200.fc31.x86_64/extra/
> Adding any weak-modules
> 
> depmod....
> 
> Backing up initramfs-5.5.17-200.fc31.x86_64.img to /boot/initramfs-5.5.17-200.fc31.x86_64.img.old-dkms
> Making new initramfs-5.5.17-200.fc31.x86_64.img
> (If next boot fails, revert to initramfs-5.5.17-200.fc31.x86_64.img.old-dkms image)
> dracut.........
> 
> DKMS: install completed.
> make[1]: Leaving directory '/home/temp/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
>  `#` ./module_operations.sh -n "" -v "" -d

Tourman36, 2020-04-21:
It looks like per the official docs, we need to create a src rpm off the current release

luis850917, 2020-04-21:
hello  tourman in fedora 30 appears me this error when try to compile using: 

make dkms

var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c: In function ‘__kfio_bind_task_to_cpumask’:
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c:129:4: error: ‘struct task_struct’ has no member named ‘cpus_mask’
  129 | tsk->cpus_mask = *mask;
      |    ^~
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c:213:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
  213 |     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
      |     ^~~~~~~~
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c:213:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
make[4]: *** [scripts/Makefile.build:277: /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:1580: _module_/var/lib/dkms/iomemory-vsl4/v5.6.0/build] Error 2
make[3]: Leaving directory '/usr/src/kernels/5.0.9-301.fc30.x86_64'
make[2]: *** [Makefile:134: modules] Error 2
make[2]: Leaving directory '/var/lib/dkms/iomemory-vsl4/v5.6.0/build'

Tourman36, 2020-04-21:
@luis850917 
This bug is for a different issue. Go see my comment in #9

closed 2020-04-21 by Tourman36