RemixVSL/iomemory-vsl4 #8: Fix dpkg and module versions to match driver

Tourman36, 2020-04-21:
fio still shows old version

>fio-status -a
Found 1 VSL driver package:
   4.3.7 build 1205 Driver: loaded
   ...

dpkg builds with old version numbers
>root@pve1:/home/temp/iomemory-vsl4# ls -a ..
.
..
deb
fio-preinstall_4.3.7.1205-1.0_amd64.deb
fio-sysvinit_4.3.7.1205-1.0_all.deb
fio-util_4.3.7.1205-1.0_amd64.deb
iomemory-vsl4
iomemory-vsl4_4.3.7.1205-1.0_amd64.buildinfo
iomemory-vsl4_4.3.7.1205-1.0_amd64.changes
iomemory-vsl4_4.3.7.1205-1.0.dsc
iomemory-vsl4_4.3.7.1205-1.0.tar.gz
iomemory-vsl4-5.3.18-3-pve_4.3.7.1205-1.0_amd64.deb
iomemory-vsl4-config-5.3.18-3-pve_4.3.7.1205-1.0_amd64.deb
iomemory-vsl4-di-5.3.18-3-pve_4.3.7.1205-1.0_amd64.udeb
iomemory-vsl4-initrd-5.3.18-3-pve_4.3.7.1205-1.0_amd64.deb
iomemory-vsl4-source_4.3.7.1205-1.0_amd64.deb

snuf, 2020-04-21:
for completeness there is a difference between the .deb file and what fio-utils reports. The .deb file is configured in ```debian/fio_values``` while fio-utils gets its version from ```libvsl_4.so``` which has the version hard coded into it. If this version matches the version the driver reports, somewhere internally the driver state is updated to loaded.

```Found 1 VSL driver package:
   4.3.6 build 1173 Driver: not loaded

Found 1 ioMemory device in this system

Adapter: ioMono  (driver 4.3.6)
	ioMemory SX300-1300, Product Number:F13-004-1300-CS-0001, SN:1446G1531, FIO SN:1446G1531
[-snip-]
```

bulgaru, 2020-09-09:
Not sure if the issues are related, but seems very likely.
If i understand the approach - the fio-util, fio-sysvinit and fio-firmware-fusion (if needed) are downloaded from the [Sandisk website](https://support.fusionio.com/downloads).

The dkms installation goes well, but when trying to install the utilities, i get the following error:

> dpkg: dependency problems prevent configuration of fio-util:
>  fio-util depends on iomemory-vsl; however:
>   Package iomemory-vsl is not installed.

This is the current setup:

> # dkms status
> iomemory-vsl, 4.3.7, 5.3.10-1-pve, x86_64: installed
> # modinfo iomemory-vsl4
> filename:       /lib/modules/5.3.10-1-pve/updates/dkms/iomemory-vsl4.ko
> version:        v5.6.0-22-ge9c4c00-4.3.7
> license:        GPL
> ...

Please advise.

snuf, 2020-09-09:
Hi @bulgaru when you install the DKMS version you don't have the dependencies dpkg will want, irrespective, as the iomemory-vsl package is not installed. You should be able to install it with dpkg by ignoring the dependencies (`--no-force-depends`).

snuf, 2022-05-17:
@bulgaru @Tourman36 I think the issue was that in older versions, not master, libvsl may not have been copied looking at the `git-blame` for `module_operations.sh` the block to do this was added 22nd of April 2020: `4c68d83d (Funs 2020-04-22 08:19:05 -0700 136) install_libvsl() {`. Closing this.

closed 2022-05-17 by snuf