RemixVSL/iomemory-vsl4 #9: Problem install fedora 30  iodrive Px600 kernel 5.0.9.300

luis850917, 2020-04-21:
### Bug description
make[2]: Entering directory '/usr/src/kernels/5.0.9-301.fc30.x86_64'
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/main.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/dbgset.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/license.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/pci.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/sysrq.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/driver_init.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kfio.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/errno.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/state.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kmsg.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/six_lock.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcache.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kfile.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kmem.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kscatter.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/ktime.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/sched.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/cdev.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kscsi.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kscsi_host.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kblock.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcondvar.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/common_kinfo.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kinfo.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kinit.o
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c: In function ‘__kfio_bind_task_to_cpumask’:
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c:129:4: error: ‘struct task_struct’ has no member named ‘cpus_mask’
  129 | tsk->cpus_mask = *mask;
      |    ^~
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c:213:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
  213 |     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
      |     ^~~~~~~~
/var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.c:213:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
make[3]: *** [scripts/Makefile.build:277: /var/lib/dkms/iomemory-vsl4/v5.6.0/build/kcpu.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1580: _module_/var/lib/dkms/iomemory-vsl4/v5.6.0/build] Error 2
make[2]: Leaving directory '/usr/src/kernels/5.0.9-301.fc30.x86_64'
make[1]: *** [Makefile:134: modules] Error 2

### How to reproduce

### Posible solution

### Environment information

Tourman36, 2020-04-21:
Can you supply the output of uname -r? I see in the official Linux kernel that cpus_mask was already there in 5.0.0. We will have to see what is included in Fedora with 5.0.9. Also, are you able to test newer kernels at all?

luis850917, 2020-04-21:
kerne 5.0.9-301.fc30 

> Can you supply the output of uname -r? I see in the official Linux kernel that cpus_mask was already there in 5.0.0. We will have to see what is included in Fedora with 5.0.9. Also, are you able to test newer kernels at all?

luis850917, 2020-04-21:
home/user/Downloads/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/ktime.c: In function ‘fusion_getwallclocktime’:
/home/user/Downloads/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/ktime.c:126:26: error: implicit declaration of function ‘current_kernel_time’; did you mean ‘core_kernel_text’? [-Werror=implicit-function-declaration]
  126 |     struct timespec ts = current_kernel_time();
      |                          ^~~~~~~~~~~~~~~~~~~
      |                          core_kernel_text
/home/user/Downloads/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/ktime.c:126:26: error: invalid initializer

Tourman36, 2020-04-21:
Thanks for the reports. I will let you know once we have fixes in place to test.

Tourman36, 2020-04-21:
@luis850917 

It looks like Fedora cherry picked mainline kernel commits into their 30 release. I just tested Fedora 31 in a VM and I was able to compile from source without errors.

snuf, 2020-04-21:
@luis850917 @Tourman36 am reading some online reports that some things should be there alright. The ```ktime_get.*``` bits are there since 4.18:
```
# flag:            KFIOC_X_HAS_COARSE_REAL_TS
# usage:           1 kernel exports ktime_get_coarse_real_ts64()
#                  0 old kernel with current_kernel_time()
# kernel version:  Added in 4.18 to provide a 64 bit time interface
#                  commit: "timekeeping: Standardize on ktime_get_*() naming"
```
The older kernels did/are supposed to/do have ```current_kernel_time()```. I do see notice of the fact that Fedora's 5.0.3 kernel should have ```ktime_get_coarse_real_ts64()```. Can you provide the output of ```root/usr/src/iomemory-vsl4-4.3.7kfio_config/KFIOC_X_HAS_COARSE_REAL_TS/kfioc_test.log``` ?

The ```cpu_mask``` missing from ```task_struct``` is a fix we removed that should have stayed. I'm going to put that back in as it breaks compatibility between 5.0 -> 5.6.

snuf, 2020-04-21:
fix for ```cpus_mask``` is in #13

luis850917, 2020-04-21:
> fix for `cpus_mask` is in #13

hello snuf, i update the kernel to 5.5.16.

when this step finished which is the next step?

Done.
/usr/src/iomemory-vsl4-v5.6.0 already exists.
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Ok:  already patched with v5.6.0-4.3.7
Adding, buidling and installing iomemory-vsl4/v5.6.0 with DKMS

Creating symlink /var/lib/dkms/iomemory-vsl4/v5.6.0/source ->
                 /usr/src/iomemory-vsl4-v5.6.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' gpl DKMS_KERNEL_VERSION=5.5.16-100.fc30.x86_64.........
cleaning build area...

DKMS: build completed.

iomemory-vsl4.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.5.16-100.fc30.x86_64/extra/
Adding any weak-modules

depmod.....

Backing up initramfs-5.5.16-100.fc30.x86_64.img to /boot/initramfs-5.5.16-100.fc30.x86_64.img.old-dkms
Making new initramfs-5.5.16-100.fc30.x86_64.img
(If next boot fails, revert to initramfs-5.5.16-100.fc30.x86_64.img.old-dkms image)
dracut............

DKMS: install completed.
make[1]: Leaving directory '/home/user/Downloads/git/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
# ./module_operations.sh -n "" -v "" -d

luis850917, 2020-04-21:
ok thank you .. now is working.. one last question. how i can install the utilities?

luis850917, 2020-04-21:
error: Failed dependencies:
	fio-util is needed by fio-sysvinit-4.3.7.1205-1.el7.x86_64

snuf, 2020-04-21:
@luis850917 good to hear it's working. The utilities you'll have to download from ```http://dell-support.sandisk.com/```, a more direct [link](https://dell.fusionio.com/downloads/fio-util-4.3.7.1205-1.el7.x86_64.rpm).

luis850917, 2020-04-21:
rpm -ivh fio-util-4.3.7.1205-1.el7.x86_64.rpm --force
error: Failed dependencies:
	iomemory-vsl is needed by fio-util-4.3.7.1205-1.el7.x86_64

snuf, 2020-04-21:
```--nodeps```, ```--force``` should work though. Otherwise you'll have to unpack the ```.rpm``` and copy the files somewhere.

luis850917, 2020-04-21:
fio-status

Error: ioMemory utilities' support library is missing.
       Make sure the ioMemory driver is properly installed and try again.

       Found 1 ioMemory card requiring v4.x driver.

snuf, 2020-04-21:
DKMS does not install ```./root/usr/lib/fio/libvsl_4.so``` which is needed for ```fio-utils``` to work

closed 2020-04-21 by Tourman36