RemixVSL/iomemory-vsl4 #22: DMKS make rpm package fails

snuf, 2020-08-17:
### Bug description
Describe the issue, or paste the full error encountered here.

### How to reproduce
What are the steps to reproduce the reported issue.

### 1st part works flawlessly
git clone https://github.com/snuf/iomemory-vsl4
cd iomemory-vsl4/
git checkout v5.7.0
Branch 'v5.7.0' set up to track remote branch 'v5.7.0' from 'origin'.
Switched to a new branch 'v5.7.0'
make dkms
[...]
DKMS: install completed.
make[1]: Leaving directory '/root/builds/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'

dkms status
iomemory-vsl4, v5.6.1-24-g9830822, 4.18.0-193.14.2.el8_2.x86_64, x86_64: installed

# ~issue fillows:
dkms mkdriverdisk -d redhat3 --media tar -m iomemory-vsl4 -v v5.6.1-24-g9830822

Using /etc/dkms/template-dkms-redhat-kmod.spec
creating source tarball...iomemory-vsl4-v5.6.1-24-g9830822/
[...]
iomemory-vsl4-v5.6.1-24-g9830822/include/fio/public/
iomemory-vsl4-v5.6.1-24-g9830822/include/fio/public/fioapi.h

rpmbuild...(bad exit status: 1)

Error! There was a problem creating your kmod.
error: line 4: Illegal char '-' (0x2d) in: Version:     v5.6.1-24-g9830822
Building target platforms: x86_64
Building for target x86_64

### Possible solution

### Environment information
Information about the system the module is used on
1.Linux drive3.localdomain 4.18.0-193.14.2.el8_2.x86_64 #1 SMP Sun Jul 26 03:54:29 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
2. gcc (GCC) 8.3.1 20191121 (Red Hat 8.3.1-5)
3. NAME="CentOS Linux"
VERSION="8 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Linux 8 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:8"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-8"
CENTOS_MANTISBT_PROJECT_VERSION="8"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="8"

4. v5.7.0
5. FIO device used, if applicable
        Dell ioDrive2 1205GB MLC, Product Number:DG1MH, SN:US0DG1MH760512810084
        External Power: NOT connected
        PCIe Power limit threshold: 24.75W
        Connected ioMemory modules:
          fct0: Product Number:DG1MH,
   * lspci -b -nn
   01:00.0 Mass storage controller [0180]: SanDisk ioDrive2 [1aed:2001] (rev 04)

snuf, 2020-08-25:
@Steven-Joles better late than never, can you try the SHA: ea8b56582e7d8e7a38c28785969e130e7d2e2a80 commit, it contains a mkrpm file and a fix for the version, if this works as expected we'll move that to master and it will be part of the next release..

@Tourman36 our versioning from a DKMS/build/compile perspective is going to move from vM.m.p-build-hash to M.m.p_build_hash, which is Red Hat "compliant", the pre-ended `v` is something `dkms mkdeb` complained about, so losing that  seems like a good idea. For the git repo it means we can leave it as is, or for clarity remove the `v` in newer releases so the tag always matches the version, any thoughts?

Tourman36, 2020-08-25:
@snuf I think it's better to have versioning be consistent across.

Steven-Joles, 2020-08-25:
@snuf 
no worries on timing, it's all good -- now if this thing would only cooperate,
is this something I'm missing on my system?
"%kernel_module_package_buildreqs" that it's searching for and I don't have installed?

##
git clone https://github.com/snuf/iomemory-vsl4
cd iomemory-vsl4
git checkout ea8b56582e7d8e7a38c28785969e130e7d2e2a80 
make dkms
    ...
    DKMS: install completed.
    
dkms status
    iomemory-vsl4, 5.6.1_24_gea8b565, 4.18.0-193.14.2.el8_2.x86_64, x86_64: installed

dkms mkdriverdisk -d redhat3 --media tar -m iomemory-vsl4 -v 5.6.1_24_gea8b565
    iomemory-vsl4-5.6.1_24_gea8b565/include/fio/public/fioapi.h
    
    rpmbuild...(bad exit status: 1)

    Error! There was a problem creating your kmod.
    error: line 10: Dependency tokens must begin with alpha-numeric, '_' or '/': BuildRequires:     %kernel_module_package_buildreqs
    Building target platforms: x86_64
    Building for target x86_64

snuf, 2020-08-27:
@Steven-Joles I was able to make the `ubuntu` driver disk work. The end result is just an iso/tgz with the kernel module in it and the sources, in places where the OS expects them when you load the driver disk in the installer. The `redhat3`  option in my case gives a very different failure result, so I can't reproduce that.

What is your intention with the driver disk, to use it for the installer? Would you be willing to try with an ubuntu iso/tgz just to see if it works (tm)?

Steven-Joles, 2020-08-27:
@snuf Absolutely,  Should be able to give it a shot within a few hours.  Let me know how you'd like to proceed.

snuf, 2022-04-14:
closing, there is an example EFI workaround for ubuntu atm.

closed 2022-04-14 by snuf