RemixVSL/iomemory-vsl4 #23: Size reported to the OS smaller than the actual?

bulgaru, 2020-09-09:
### Bug description

> $ fio-status -a
> 
> Found 1 VSL driver package:
>    4.3.7 build 1205 Driver: loaded
> 
> Found 1 ioMemory device in this system
> 
> Adapter: ioMono  (driver 4.3.7)
>         ioMemory SX350-3200, Product Number:SDFADAMOS-3T20-SF1, SN:1711D01B7, FIO SN:1711D01B7
>         ioMemory Adapter Controller, PN:5491-72374-3200G
>         Product UUID:d38c7ed4-874a-5e7a-abf7-9117a2171f33
>         PCIe Bus voltage: avg 12.21V
>         PCIe Bus current: avg 0.64A
>         PCIe Bus power: avg 7.83W
>         PCIe Power limit threshold: 24.75W
>         PCIe slot available power: 25.00W
>         PCIe negotiated link: 8 lanes at 5.0 Gt/sec each, 4000.00 MBytes/sec total
>         Connected ioMemory modules:
>           fct0: 03:00.0,        Product Number:SDFADAMOS-3T20-SF1, SN:1711D01B7
> 
> fct0    Attached
>         ioMemory Adapter Controller, Product Number:SDFADAMOS-3T20-SF1, SN:1711D01B7
>         ioMemory Adapter Controller, PN:5491-72374-3200G
>         Microcode Versions: App:0.0.15.0
>         Powerloss protection: protected
>         PCI:03:00.0, Slot Number:2
>         Vendor:1aed, Device:3002, Sub vendor:1aed, Sub device:3002
>         Firmware v8.9.9, rev 20190313 Public
>         2560.00 GBytes device size
>         Format: v501, 625000000 sectors of 4096 bytes
>         PCIe slot available power: 25.00W
>         PCIe negotiated link: 8 lanes at 5.0 Gt/sec each, 4000.00 MBytes/sec total
>         Internal temperature: 43.31 degC, max 48.72 degC
>         Internal voltage: avg 1.01V, max 1.02V
>         Aux voltage: avg 1.78V, max 1.80V
>         Reserve space status: Healthy; Reserves: 100.00%, warn at 10.00%
>         Active media: 100.00%
>         Rated PBW: 11.00 PB, 99.83% remaining
>         Lifetime data volumes:
>            Physical bytes written: 18,225,007,081,504
>            Physical bytes read   : 9,757,190,920,192
>         RAM usage:
>            Current: 274,637,696 bytes
>            Peak   : 274,637,696 bytes
>         Contained Virtual Partitions:
>           fioa: ID:0, UUID:9a9d4471-91e3-924a-8224-b1e2919a9b82
> 
> fioa    State: Online, Type: block device, Device: /dev/fioa
>         ID:0, UUID:9a9d4471-91e3-924a-8224-b1e2919a9b82
>         2560.00 GBytes device size
>         Format: 625000000 sectors of 4096 bytes
>         Sectors In Use: 70
>         Max Physical Sectors Allowed: 625000000
>         Min Physical Sectors Reserved: 625000000

Notice the 2560.00 GBytes device size.

### How to reproduce
Used the guide from here: https://forum.proxmox.com/threads/configuring-fusion-io-sandisk-sx300-sx350-and-px600-cards-with-proxmox-6-1-kernel-5-x-tested-up-to-5-6.68642/

I'm using SDFADAMOS-3T20-SF1

### Possible solution
I assume there is a hard limit of how big the drive can be and 3.2TB exceeds it.

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a): Linux dc6 5.4.60-1-pve #1 SMP PVE 5.4.60-2 (Fri, 04 Sep 2020 10:24:50 +0200) x86_64 GNU/Linux
2. The C compiler version used (gcc --version): gcc (Debian 8.3.0-6) 8.3.0
3. distribution, and version (cat /etc/os-release): Debian GNU/Linux 10 (buster)
4. Tag or Branch of iomemory-vsl4 that is being compiled: v5.6.1

snuf, 2020-09-09:
The drive can be formatted with reserve space for performance reasons. Chances are your card has been formatted to a reduced capacity. In your case this would probably be ~25%, as that would add up from 2.5TB to 3.2TB. Please have a look at page 31 [here](https://support.fusionio.com/load/-media-/2fk40u/docsConfluence/ioMemory_VSL_Peak_Performance_Guide_2013-08-20.pdf)

bulgaru, 2020-09-09:
Thank you, @snuf!

As always, you're very helpful! After some digging i arrive to the same conclusion. 
Glad the drive is ok and glad the drivers seem to work fine! :)

closed 2020-09-09 by bulgaru