RemixVSL/iomemory-vsl4 #24: Missing files when compiling?

juchong, 2020-10-03:
### Bug description
Compilation issue on unRAID (Slackware)

### How to reproduce
What are the steps to reproduce the reported issue.
```
git clone https://github.com/snuf/iomemory-vsl4.git
cd /iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/
git checkout 86d697c246677a4f5c95c04f714a4fabbdf56070
make
```
Here's what I get when trying to compile:
```
root@Tower:~/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7# make gpl
sed -i 's/Proprietary/GPL/g' license.c
if [ "92" -gt "92" ];then \
        if [ ! -f "kfio/x86_64_cc92_libkfio.o_shipped" ];then \
                cp kfio/x86_64_cc92_libkfio.o_shipped kfio/x86_64_cc92_libkfio.o_shipped; \
        fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/4.19.107-Unraid/build -p -d /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config -l 0 -s /lib/modules/4.19.107-Unraid/source
Detecting Kernel Flags
Config dir         : /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/4.19.107-Unraid/build
Kernel source dir  : /lib/modules/4.19.107-Unraid/source
Starting tests:
  1601746574.718  KFIOC_HAS_GLOBAL_REGS_POINTER...
  1601746574.724  KFIOC_HAS_SYSRQ_KEY_OP_ENABLE_MASK...
  1601746574.731  KFIOC_FOPS_USE_LOCKED_IOCTL...
  1601746574.738  KFIOC_HAS_RQ_POS_BYTES...
  1601746574.747  KFIOC_NUMA_MAPS...
  1601746574.755  KFIOC_HAS_SCSI_QD_CHANGE_FN...
  1601746574.764  KFIOC_X_HAS_COARSE_REAL_TS...
  1601746574.772  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS...
  1601746574.783  KFIOC_X_TASK_HAS_CPUS_MASK...
Started tests, waiting for completions...
  1601746576.844  KFIOC_HAS_GLOBAL_REGS_POINTER=0
  1601746576.871  KFIOC_HAS_SYSRQ_KEY_OP_ENABLE_MASK=0
  1601746576.898  KFIOC_FOPS_USE_LOCKED_IOCTL=0
  1601746576.924  KFIOC_HAS_RQ_POS_BYTES=0
  1601746576.951  KFIOC_NUMA_MAPS=0
  1601746576.978  KFIOC_HAS_SCSI_QD_CHANGE_FN=0
  1601746577.004  KFIOC_X_HAS_COARSE_REAL_TS=0
  1601746577.031  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS=0
  1601746577.058  KFIOC_X_TASK_HAS_CPUS_MASK=0
Finished
1601746577.071  Exiting
Preserving configdir due to '-p' option: /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config

Adding module version v5.6.1-4.3.7 to source license.c

make \
    -j8 \
-C /lib/modules/4.19.107-Unraid/build \
FIO_DRIVER_NAME=iomemory-vsl4 \
FUSION_DRIVER_DIR=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
M=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
EXTRA_CFLAGS+="-I/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall" \
INSTALL_MOD_DIR=extra/fio \
INSTALL_MOD_PATH= \
KFIO_LIB=kfio/x86_64_cc92_libkfio.o_shipped \
modules
make[1]: Entering directory '/usr/src/linux-4.19.107-Unraid'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/main.o
  CC [M]  /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/dbgset.o
[... 82 lines cut ...]
                 from ./include/linux/smp.h:64,
                 from ./include/linux/percpu.h:7,
                 from ./include/linux/context_tracking_state.h:5,
                 from ./include/linux/vtime.h:5,
                 from ./include/linux/hardirq.h:8,
                 from ./include/linux/interrupt.h:11,
                 from /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/pci.c:30:
./arch/x86/include/asm/io.h:44:10: fatal error: asm/early_ioremap.h: No such file or directory
   44 | #include <asm/early_ioremap.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:303: /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/pci.o] Error 1
make[1]: *** [Makefile:1522: _module_/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7] Error 2
make[1]: Leaving directory '/usr/src/linux-4.19.107-Unraid'
make: *** [Makefile:127: modules] Error 2
```
### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a): **4.19.107**
2. The C compiler version used (gcc --version)**v5.3**
3. distribution, and version (cat /etc/os-release) **unRAID (Slackware)**
4. Tag or Branch of iomemory-vsl4 that is being compiled **master**
5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn

snuf, 2020-10-03:
Hi @juchong thanks for the report, unfortunately the main branch does NOT support kernels lower than 5.X. You can try using the 7e12eef branch, which is a modified version of the original driver, otherwise the original driver should support the 4.19 kernel.

juchong, 2020-10-03:
Thanks, @snuf! Let me give that a try.

juchong, 2020-10-03:
It looks like I'm still missing `bounds.h` for some reason, even when switching over to the vanilla branch. Let me see if I can't figure out why. Please let me know if you have any ideas!

snuf, 2020-10-03:
@juchong seems like you're missing the kernel header files?

juchong, 2020-10-03:
That was exactly the problem! It looks like the build completed successfully, too. Sorry for all of the basic questions!

snuf, 2020-10-03:
No worries, happy it worked out! Shall I close this issue?

juchong, 2020-10-03:
I've got it, no worries.

closed 2020-10-03 by juchong