RemixVSL/iomemory-vsl4 #25: Block Allocation Queue Uses Deprecated Functions

juchong, 2020-10-13:
### Bug description
I'm trying to compile the driver on a slightly newer kernel (5.8.13) and have hammered out most of the issues, except for the block queue requests. 

NOTE: This might better fit a feature request. If so, feel free to move it over. 

### How to reproduce
NOTE: I've made a few modifications to the driver on my own fork. 
```
clone https://github.com/juchong/iomemory-vsl4.git
cd iomemory-vsl4
git checkout dkms_mkdeb_mkrpm
make dkms
```
Error Log:
```
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
  214 |     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
      |     ^~~~~~~~
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
  CC [M]  /var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kcsr.o
  SHIPPED /var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kfio/x86_64_cc102_libkfio.o
  CC [M]  /var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/module_param.o
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kblock.c: In function ‘kfio_alloc_queue’:
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kblock.c:1600:10: error: implicit declaration of function ‘blk_alloc_queue_node’; did you mean ‘blk_alloc_queue’? [-Werror=implicit-function-declaration]
 1600 |     rq = blk_alloc_queue_node(GFP_NOIO, node);
      |          ^~~~~~~~~~~~~~~~~~~~
      |          blk_alloc_queue
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kblock.c:1600:8: warning: assignment to ‘struct request_queue *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 1600 |     rq = blk_alloc_queue_node(GFP_NOIO, node);
      |        ^
/var/lib/dkms/iomemory-vsl4/5.6.1_29_g2e84756/build/kblock.c:1604:9: error: implicit declaration of function ‘blk_queue_make_request’; did you mean ‘blk_mq_make_request’? [-Werror=implicit-function-declaration]
 1604 |         blk_queue_make_request(rq, kfio_make_request);
      |         ^~~~~~~~~~~~~~~~~~~~~~
      |         blk_mq_make_request
cc1: some warnings being treated as errors
```
### Possible solution
It looks like the block functions were re-worked in 5.8.x. More info [here](https://prog.world/linux-kernel-5-0-we-write-simple-block-device-under-blk-mq/).

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against: 5.8.13-Unraid
2. The C compiler version used: Debian 10.2.0-9
3. Distribution and version: Slackware 14.2 x86_64
4. Tag or Branch of iomemory-vsl4 that is being compiled: dkms_mkdeb_mkrpm
5. FIO device used, if applicable: OEM ioMemory SX350 6.4TB

juchong, 2020-10-14:
The latest commit in the `dkms_mkdeb_mkrpm` branch seems to be happy on 5.8.13-Unraid! @snuf do you think you could share your test procedure/scripts for me to verify on my end?

snuf, 2020-10-14:
@juchong hey, thanks for filing and your patience! I think I had some fixes staged for 5.8.0 on my end here, but they seem to have been for iomemory-vsl, not vls4. The test scripts are here: https://github.com/snuf/hakmotif, specifically in module_test. which relies on the envfile in the root, ideally I'd like to rewrite the shell parts in Ansible or something else at some point).

snuf, 2020-10-14:
@juchong the blk_queue_make_request bugs were solved in the v5.7.0 branch, there is however an issue left for 5.8.x related to mmap. This has been fixed in iomemory-vsl, but I'm not happy about the fix, as there is a cleaner way of doing it, this is the reason it didn't make it into iomemory-vsl4 yet:

juchong, 2020-10-14:
Sounds good! I've managed to get things working on my end by changing the driver to use `blk_alloc_queue` but ended up completely eliminating `mmap_sem`. https://github.com/juchong/iomemory-vsl4/commit/347f6711be61fba167cd7462bb68cb2b88fc8ea7

snuf, 2020-10-14:
Ah I see what you've done, that might break backwards compatibility I think though, I have to double check that. The driver needs to be able to go from 5.0 > now. I solved it like this: https://github.com/snuf/iomemory-vsl4/commit/3086219155b00fa44bc7410bb07fd9ed40b96b66,

juchong, 2020-10-16:
I rolled all of your changes into 08b4e4d4a9b712845c69507c765ec9133a14f63e but left some of the `dkms` fixes in place to keep everything happy. I haven't had a chance to set a machine up for running tests, but the card seems to behave well under "normal" use.

snuf, 2020-10-16:
Nice, sorry for the tardy response on my side, life kind of gets in the way of things sometimes! The DKMS fixes were specifically for working with building driver disks, this is if you'd want to boot from your UEFI capable FIO device.

juchong, 2020-10-17:
No worries! I don't need that for my specific purpose, but it's still good to have. Now that the driver seems happy, I'm going to try and make a few tweaks to the UDEV rules to try and make Unraid detect these ioMemory devices as NVME drives.

snuf, 2021-01-09:
Fixes for this are all in master now, including 5.9.x support. There is also a new udev rules file which lives in tools. Closing this reminder @juchong again thanks for the elaborate report!

closed 2021-01-09 by snuf