RemixVSL/iomemory-vsl4 #29: failing compiling on proxmox 5.11.7-1-pve

nhidog, 2021-05-04:
all sorts of compilation errors 5.11.7-1-pve0.  compiles fine on 5.4.106-1-pve.  unfortunately trying to get to 5.11 to take care of my networking issue.

/root/foo/iomemory-vsl4/driver_source/kfile.c: In function ‘kfio_create_proc_fops_entry’:
/root/foo/iomemory-vsl4/driver_source/kfile.c:225:47: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
     return proc_create_data(name, mode, base, (struct file_operations *)fops, data);
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/foo/iomemory-vsl4/driver_source/kfile.c:37:
./include/linux/proc_fs.h:104:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘struct file_operations *’
 extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
                               ^~~~~~~~~~~~~~~~
/root/foo/iomemory-vsl4/driver_source/kcpu.c: In function ‘__kfio_bind_task_to_cpumask’:
/root/foo/iomemory-vsl4/driver_source/kcpu.c:131:10: error: ‘struct task_struct’ has no member named ‘cpus_allowed’; did you mean ‘nr_cpus_allowed’?
     tsk->cpus_allowed = *mask;
          ^~~~~~~~~~~~
          nr_cpus_allowed
/root/foo/iomemory-vsl4/driver_source/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/root/foo/iomemory-vsl4/driver_source/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
     ^~~~~~~~
/root/foo/iomemory-vsl4/driver_source/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
make[4]: *** [scripts/Makefile.build:287: /root/foo/iomemory-vsl4/driver_source/kcpu.o] Error 1
make[4]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:287: /root/foo/iomemory-vsl4/driver_source/kfile.o] Error 1
/root/foo/iomemory-vsl4/driver_source/kblock.c: In function ‘linux_bdev_update_stats’:
/root/foo/iomemory-vsl4/driver_source/kblock.c:959:9: error: implicit declaration of function ‘part_stat_lock’; did you mean ‘virt_spin_lock’? [-Werror=implicit-function-declaration]
         part_stat_lock();
         ^~~~~~~~~~~~~~
         virt_spin_lock
/root/foo/iomemory-vsl4/driver_source/kblock.c:960:9: error: implicit declaration of function ‘part_stat_inc’; did you mean ‘param_set_int’? [-Werror=implicit-function-declaration]
         part_stat_inc(GD_PART0, ios[1]);
         ^~~~~~~~~~~~~
         param_set_int
/root/foo/iomemory-vsl4/driver_source/kblock.c:960:33: error: ‘ios’ undeclared (first use in this function); did you mean ‘insl’?
         part_stat_inc(GD_PART0, ios[1]);
                                 ^~~
                                 insl
/root/foo/iomemory-vsl4/driver_source/kblock.c:960:33: note: each undeclared identifier is reported only once for each function it appears in
/root/foo/iomemory-vsl4/driver_source/kblock.c:961:9: error: implicit declaration of function ‘part_stat_add’; did you mean ‘page_ref_add’? [-Werror=implicit-function-declaration]
         part_stat_add(GD_PART0, sectors[1], totalsize >> 9);
         ^~~~~~~~~~~~~
         page_ref_add
/root/foo/iomemory-vsl4/driver_source/kblock.c:961:33: error: ‘sectors’ undeclared (first use in this function); did you mean ‘sector_t’?
         part_stat_add(GD_PART0, sectors[1], totalsize >> 9);
                                 ^~~~~~~
                                 sector_t
/root/foo/iomemory-vsl4/driver_source/kblock.c:962:33: error: ‘nsecs’ undeclared (first use in this function)
         part_stat_add(GD_PART0, nsecs[1], kfio_div64_64(duration * HZ, FIO_USEC_PER_SEC));
                                 ^~~~~
/root/foo/iomemory-vsl4/driver_source/kblock.c:963:9: error: implicit declaration of function ‘part_stat_unlock’; did you mean ‘bit_spin_unlock’? [-Werror=implicit-function-declaration]
         part_stat_unlock();
         ^~~~~~~~~~~~~~~~
         bit_spin_unlock
/root/foo/iomemory-vsl4/driver_source/kblock.c: In function ‘linux_bdev_update_inflight’:
/root/foo/iomemory-vsl4/driver_source/kblock.c:994:9: error: implicit declaration of function ‘part_stat_set_all’; did you mean ‘part_start_io_acct’? [-Werror=implicit-function-declaration]
         part_stat_set_all(GD_PART0, in_flight);
         ^~~~~~~~~~~~~~~~~
         part_start_io_acct
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:287: /root/foo/iomemory-vsl4/driver_source/kblock.o] Error 1
make[3]: *** [Makefile:1834: /root/foo/iomemory-vsl4/driver_source] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make[2]: *** [Makefile:134: modules] Error 2
make[2]: Leaving directory '/root/foo/iomemory-vsl4/driver_source'
ERROR:
make[1]: *** [debian/rules:102: build-arch-stamp] Error 1
make[1]: Leaving directory '/root/foo/iomemory-vsl4'
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make: *** [Makefile:14: dpkg] Error 2

snuf, 2021-05-04:
Hi @nhidog, thanks for filing a bug. Which branch/tag are you running?

nhidog, 2021-05-04:
v5.12.0

snuf, 2021-05-04:
can you share the `git branch -v` ?

nhidog, 2021-05-04:
```
root@t5810:~/foo/iomemory-vsl4# git branch -v
* (HEAD detached at v5.12.0) b76f1dc Update README.md
  master                     2741d03 Update README.md
root@t5810:~/foo/iomemory-vsl4# 
```

snuf, 2021-05-04:
The source for v5.12.0 has been tested to compile and load on 5.4.0, 5.8.0, 5.11.0 and .5.12.0. Can you do a clean checkout, and do a make module from the iomemory.vsl4 dir and put the output here? Am curious if it's the dpkg making that is doing something weird.

nhidog, 2021-05-04:
created new vm to instead of my host :)

apt install pve-kernel-5.11 pve-headers-5.11
reboot
apt install git nano screen patch fakeroot build-essential devscripts libncurses5 libncurses5-dev libssl-dev bc flex bison libelf-dev libaudit-dev libgtk2.0-dev libperl-dev asciidoc xmlto gnupg gnupg2 rsync lintian debhelper libdw-dev libnuma-dev libslang2-dev sphinx-common asciidoc-base automake cpio dh-python file gcc kmod libiberty-dev libpve-common-perl libtool perl-modules python-minimal sed tar zlib1g-dev lz4

`root@pve:~/iomemory-vsl4# git checkout v5.12.0 
Note: checking out 'v5.12.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at b76f1dc Update README.md
root@pve:~/iomemory-vsl4# make module
cd root/usr/src/iomemory-vsl4-4.3.7 && \
        make clean
make[1]: Entering directory '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make \
        -j4 \
        -C /lib/modules/5.11.7-1-pve/build \
        FIO_DRIVER_NAME=iomemory-vsl4 \
        FUSION_DRIVER_DIR=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
        M=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
        EXTRA_CFLAGS+="-I/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall" \
        KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
        clean
make[2]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[2]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[1]: Leaving directory '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
cd root/usr/src/iomemory-vsl4-4.3.7 && \
        make gpl
make[1]: Entering directory '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
sed -i 's/Proprietary/GPL/g' Kbuild

Change found in target kernel: KERNELVER KERNEL_SRC 
Running clean before building driver

make[2]: Entering directory '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make \
        -j4 \
        -C /lib/modules/5.11.7-1-pve/build \
        FIO_DRIVER_NAME=iomemory-vsl4 \
        FUSION_DRIVER_DIR=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
        M=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
        EXTRA_CFLAGS+="-I/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall" \
        KFIO_LIB=kfio/x86_64_cc83_libkfio.o_shipped \
        clean
make[3]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[3]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[2]: Leaving directory '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
if [ "83" -gt "75" ];then \
        if [ ! -f "kfio/x86_64_cc83_libkfio.o_shipped" ];then \
                cp kfio/x86_64_cc75_libkfio.o_shipped kfio/x86_64_cc83_libkfio.o_shipped; \
        fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/5.11.7-1-pve/build -p -d /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config -l 0 -s /lib/modules/5.11.7-1-pve/source
Detecting Kernel Flags
Config dir         : /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/5.11.7-1-pve/build
Kernel source dir  : /lib/modules/5.11.7-1-pve/build
Starting tests:
[... 190 lines cut ...]
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kblock.c: In function ‘linux_bdev_update_inflight’:
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kblock.c:994:9: error: implicit declaration of function ‘part_stat_set_all’; did you mean ‘part_start_io_acct’? [-Werror=implicit-function-declaration]
         part_stat_set_all(GD_PART0, in_flight);
         ^~~~~~~~~~~~~~~~~
         part_start_io_acct
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:287: /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kblock.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c: In function ‘__kfio_bind_task_to_cpumask’:
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c:131:10: error: ‘struct task_struct’ has no member named ‘cpus_allowed’; did you mean ‘nr_cpus_allowed’?
     tsk->cpus_allowed = *mask;
          ^~~~~~~~~~~~
          nr_cpus_allowed
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
     ^~~~~~~~
/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
make[3]: *** [scripts/Makefile.build:287: /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.o] Error 1
make[2]: *** [Makefile:1834: /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make[1]: *** [Makefile:134: modules] Error 2
make[1]: Leaving directory '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make: *** [Makefile:29: module] Error 2
root@pve:~/iomemory-vsl4# `

snuf, 2021-05-04:
it looks like the tests are failing on your host(s). The first "error", should not be reached as there is a test that checks to see what `PROC_CREATE_DATA` wants,:
``` froot/usr/src/iomemory-vsl4-4.3.7/kfile.c
224 #if ! KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS$
225     return proc_create_data(name, mode, base, (struct file_operations *)fops, data);$
226 #elif KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS$
227     return proc_create_data(name, mode, base, (struct proc_ops *)fops, data);$
228 #endif /* KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS */$
```
The error indicates that the test says it doesn't want PROC_OPS, which it in actuality does. Looking at the rest of the errors it seems like the kernel headers are incomplete and missing the base headers. Can you got to `root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS` and type `make` and parse the output here to verify?

I assume this is proxmox 6.4-1?

nhidog, 2021-05-04:
6.3-1 since i had the iso on hand

```root@pve:~/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS# make
make -C /lib/modules/5.11.7-1-pve/build M=/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS EXTRA_CFLAGS='-Wall -Werror-implicit-function-declaration' modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.ko'.  Stop.
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1738: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make: *** [Makefile:12: modules] Error 2
root@pve:~/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS# pveversion --verbose
proxmox-ve: 6.3-1 (running kernel: 5.11.7-1-pve)
pve-manager: 6.3-2 (running version: 6.3-2/22f57405)
pve-kernel-5.11: 7.0-0+3~bpo10
pve-kernel-5.4: 6.3-1
pve-kernel-helper: 6.3-1
pve-kernel-5.11.7-1-pve: 5.11.7-1~bpo10
pve-kernel-5.4.73-1-pve: 5.4.73-1
ceph-fuse: 12.2.11+dfsg1-2.1+b1
corosync: 3.0.4-pve1
criu: 3.11-3
glusterfs-client: 5.5-3
ifupdown: 0.8.35+pve1
ksm-control-daemon: 1.3-1
libjs-extjs: 6.0.1-10
libknet1: 1.16-pve1
libproxmox-acme-perl: 1.0.5
libproxmox-backup-qemu0: 1.0.2-1
libpve-access-control: 6.1-3
libpve-apiclient-perl: 3.0-3
libpve-common-perl: 6.4-2
libpve-guest-common-perl: 3.1-3
libpve-http-server-perl: 3.0-6
libpve-storage-perl: 6.3-1
libqb0: 1.0.5-1
libspice-server1: 0.14.2-4~pve6+1
lvm2: 2.03.02-pve4
lxc-pve: 4.0.3-1
lxcfs: 4.0.3-pve3
novnc-pve: 1.1.0-1
proxmox-backup-client: 1.0.5-1
proxmox-mini-journalreader: 1.1-1
proxmox-widget-toolkit: 2.4-3
pve-cluster: 6.2-1
pve-container: 3.3-1
pve-docs: 6.3-1
pve-edk2-firmware: 2.20200531-1
pve-firewall: 4.1-3
pve-firmware: 3.1-3
pve-ha-manager: 3.1-1
pve-i18n: 2.2-2
pve-qemu-kvm: 5.1.0-7
pve-xtermjs: 4.7.0-3
qemu-server: 6.3-1
smartmontools: 7.1-pve2
spiceterm: 3.1-1
vncterm: 1.6-2
zfsutils-linux: 0.8.5-pve1
root@pve:~/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS#```

snuf, 2021-05-04:
Looks like your headers are missing some bits alright:
```
make[1]: Entering directory '/usr/src/linux-headers-5.11.7-1-pve'
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.ko'.  Stop.
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1738: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.7-1-pve'
make: *** [Makefile:12: modules] Error 2
```
Is there another package you're missing to be able to compile kernel modules on proxmox?

snuf, 2021-05-04:
Normally the `scripts/module.lds` lives in the `linux-headers-<version>` file:
```
snuf@scipio:/var/lib/dpkg/info$ grep scripts/module.lds *.list
linux-headers-5.11.0-16-generic.list:/usr/src/linux-headers-5.11.0-16-generic/scripts/module.lds
linux-headers-5.11.0-16-generic.list:/usr/src/linux-headers-5.11.0-16-generic/scripts/module.lds.S
linux-headers-5.11.0-16.list:/usr/src/linux-headers-5.11.0-16/scripts/module.lds.S
```

nhidog, 2021-05-04:
thanks for pointing me in the right direction found out it was missng module.lds, after copying was able to build, alright hopefully it all just works
```
cp /usr/src/linux-headers-5.11.7-1-pve/scripts/module.lds.S /usr/src/linux-headers-5.11.7-1-pve/scripts/module.lds
```

closed 2021-05-04 by nhidog