RemixVSL/iomemory-vsl4 #32: DKMS fails to build on 5.12.4

4nanook, 2021-05-15:
### Bug description
Describe the issue, or paste the full error encountered here.
Trying to update kernel from 5.10.35 to 5.12.4, getting the following error:
Setting up linux-image-5.12.4 (5.12.4-1) ...
 * dkms: running auto installation service for kernel 5.12.4                    Error! Could not locate dkms.conf file.
File: /var/lib/dkms/iomemory-vsl/v5.10.0/source/dkms.conf does not exist.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' DKMS_KERNEL_VERSION=5.12.4...(bad exit status: 2)
ERROR (dkms apport): binary package for iomemory-vsl4: 5.10.35-e2084ae not found
Error! Bad return status for module build on kernel: 5.12.4 (x86_64)
Consult /var/lib/dkms/iomemory-vsl4/5.10.35-e2084ae/build/make.log for more information.
(Log attached below)
[tmp.make.log](https://github.com/snuf/iomemory-vsl4/files/6481640/tmp.make.log)

### How to reproduce
What are the steps to reproduce the reported issue.
```
Built kernel using make -j8 bindeb-pkg and installed *.deb files with dpkg -i *.deb
git clone https://github.com/snuf/iomemory-vsl4.git
cd iomemory-vsl4
git checkout <tag or some-branch>
make module
** poof, broken token **
```

### Possible solution
Is a solution know, or type any plausible suggestions here, if none leave clear.

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a)
2. The C compiler version used (gcc --version)
3. distribution, and version (cat /etc/os-release)
4. Tag or Branch of iomemory-vsl4 that is being compiled
5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn

snuf, 2021-05-15:
Hi @4nanook 

The following error:
```
/var/lib/dkms/iomemory-vsl4/5.10.35-e2084ae/build/kfile.c:225:47: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  225 |     return proc_create_data(name, mode, base, (struct file_operations *)fops, data);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               struct file_operations *
In file included from /var/lib/dkms/iomemory-vsl4/5.10.35-e2084ae/build/kfile.c:37:
./include/linux/proc_fs.h:104:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘struct file_operations *’
  104 | extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
      |                               ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```
Seems to indicate there is something wrong with the headers, besides the GLIBC issue that it's spitting out in the logs. The test it is failing on is KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS. I advise you do a `make module` from the module directory itself and investigate why the `kfio/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS` test is failing on your system.

4nanook, 2021-05-15:
This is what I get:
make modules
  SYNC    include/config/auto.conf.cmd
  HOSTCC  scripts/kconfig/conf.o
scripts/basic/fixdep: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by scripts/basic/fixdep)
make[2]: *** [scripts/Makefile.host:112: scripts/kconfig/conf.o] Error 1
make[2]: *** Deleting file 'scripts/kconfig/conf.o'
make[1]: *** [Makefile:603: syncconfig] Error 2
make: *** [Makefile:711: include/config/auto.conf.cmd] Error 2
make: *** [include/config/auto.conf.cmd] Deleting file 'include/generated/autoconf.h'
The version glibc that is present on Ubuntu 20.04 is 2.31

snuf, 2021-05-15:
are you typing `make module` or `make modules`, and where in the path?

4nanook, 2021-05-15:
Sorry, I typed make modules, but here is the output from "make module":
root@iglulik:/lib/modules/5.10.37/build# make module
  SYNC    include/config/auto.conf.cmd
  HOSTCC  scripts/kconfig/conf.o
scripts/basic/fixdep: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by scripts/basic/fixdep)
make[2]: *** [scripts/Makefile.host:112: scripts/kconfig/conf.o] Error 1
make[2]: *** Deleting file 'scripts/kconfig/conf.o'
make[1]: *** [Makefile:603: syncconfig] Error 2
make: *** [Makefile:711: include/config/auto.conf.cmd] Error 2
root@iglulik:/lib/modules/5.10.37/build# I am in /lib/modules/5.10.37/build

4nanook, 2021-05-15:
Ok I think I misunderstood, I was trying to get dkms to work, building in /usr/src/iomemory-vls4 I get:
cd root/usr/src/iomemory-vsl4-4.3.7 && \
  	make clean
make[1]: Entering directory '/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make \
        -j12 \
	-C /lib/modules/5.10.35/build \
	FIO_DRIVER_NAME=iomemory-vsl4 \
	FUSION_DRIVER_DIR=/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
	M=/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
	EXTRA_CFLAGS+="-I/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall" \
	KFIO_LIB=kfio/x86_64_cc93_libkfio.o_shipped \
	clean
make[2]: Entering directory '/usr/src/linux-headers-5.10.35'
make[2]: Leaving directory '/usr/src/linux-headers-5.10.35'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[1]: Leaving directory '/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
cd root/usr/src/iomemory-vsl4-4.3.7 && \
  	make gpl
make[1]: Entering directory '/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
sed -i 's/Proprietary/GPL/g' Kbuild

Change found in target kernel: KERNELVER KERNEL_SRC 
Running clean before building driver

make[2]: Entering directory '/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make \
        -j12 \
	-C /lib/modules/5.10.35/build \
	FIO_DRIVER_NAME=iomemory-vsl4 \
	FUSION_DRIVER_DIR=/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
	M=/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7 \
	EXTRA_CFLAGS+="-I/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/include -DBUILDING_MODULE -DLINUX_IO_SCHED -Wall" \
	KFIO_LIB=kfio/x86_64_cc93_libkfio.o_shipped \
	clean
make[3]: Entering directory '/usr/src/linux-headers-5.10.35'
make[3]: Leaving directory '/usr/src/linux-headers-5.10.35'
rm -rf include/fio/port/linux/kfio_config.h kfio_config license.c
make[2]: Leaving directory '/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
if [ "93" -gt "75" ];then \
	if [ ! -f "kfio/x86_64_cc93_libkfio.o_shipped" ];then \
		cp kfio/x86_64_cc75_libkfio.o_shipped kfio/x86_64_cc93_libkfio.o_shipped; \
	fi \
fi
./kfio_config.sh -a x86_64 -o include/fio/port/linux/kfio_config.h -k /lib/modules/5.10.35/build -p -d /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config -l 0 -s /lib/modules/5.10.35/source
Detecting Kernel Flags
Config dir         : /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config
Output file        : include/fio/port/linux/kfio_config.h
Kernel output dir  : /lib/modules/5.10.35/build
Kernel source dir  : /lib/modules/5.10.35/build
Starting tests:
  1621043055.760  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS...
  1621043055.777  KFIOC_X_TASK_HAS_CPUS_MASK...
  1621043055.796  KFIOC_X_LINUX_HAS_PART_STAT_H...
  1621043055.812  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS...
  1621043055.829  KFIOC_X_HAS_MAKE_REQUEST_FN...
  1621043055.850  KFIOC_X_GENHD_PART0_IS_A_POINTER...
  1621043055.868  KFIOC_X_BIO_HAS_BI_BDEV...
Started tests, waiting for completions...
  1621043057.937  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS=1
  1621043057.950  KFIOC_X_TASK_HAS_CPUS_MASK=1
  1621043057.962  KFIOC_X_LINUX_HAS_PART_STAT_H=1
  1621043057.989  KFIOC_X_BLK_ALLOC_QUEUE_NODE_EXISTS=0
  1621043058.000  KFIOC_X_HAS_MAKE_REQUEST_FN=0
  1621043058.039  KFIOC_X_GENHD_PART0_IS_A_POINTER=0
  1621043058.051  KFIOC_X_BIO_HAS_BI_BDEV=0
Finished
1621043058.057  Exiting
Preserving configdir due to '-p' option: /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config
grep: license.c: No such file or directory
[... 36 lines cut ...]
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcondvar.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/common_kinfo.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kinfo.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kinit.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_common.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/khotplug.o
/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
  214 |     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
      |     ^~~~~~~~
/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kcsr.o
  SHIPPED /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio/x86_64_cc93_libkfio.o
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/module_param.o
  LD [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.o
  MODPOST /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/Module.symvers
  CC [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.mod.o
  LD [M]  /usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.ko
make[2]: Leaving directory '/usr/src/linux-headers-5.10.35'

Ok: patched version for iomemory-vsl4.ko replaced 4.3.7.1205 bryce@a7e621e6fb50 with 5.10.35-e2084ae-4.3.7.1205

make[1]: Leaving directory '/usr/src/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
Which works but 5.10.35 is the existing kernel, I need it to work for 5.10.37 or better yet 5.12.4

snuf, 2021-05-15:
Kay, so one thing at a time. 
1) Your headers for 5.12.4 are not complete I think 
2) Do you have the headers installed for 5.10.37?. If you don't it will not compile.
3) The GLIBC issue is well an issue, have you tried booting into your new kernel and running the dkms install?

4nanook, 2021-05-15:
No I haven't, it is in service now and traffic too high this time of day.  I'm reasonably sure the headers are intact.  I built kernel with make -j8 bindeb-pkg and dpkg -i the resulting .deb files BUT I did not build on same computer.  The computer I built the kernel on does have libc6 installed because it is running Ubuntu 21.04, I do not want to upgrade this machine to 21.04 because I depend upon some PPA's that are not maintained.  Perhaps I need to build the kernel on that machine (a very tedious process, it is not as snappy as my build machine) so libs match? I will give that a try and see if it gets happier.

4nanook, 2021-05-15:
Ok, I built on a 21.04 machine, that is why glibc mismatch exists.  I guess I can build kernel on this machine but is slow and painful.

4nanook, 2021-05-15:
Ok with the kernel built on that machine the module builds without errors.

snuf, 2021-05-15:
@4nanook happy it worked out, can we close this one?

4nanook, 2021-05-15:
Yep, all up and running on 5.12.4, just needed to build kernel on that machine so libc6 had correct version.

closed 2021-05-15 by 4nanook