RemixVSL/iomemory-vsl4 #33: Failed to compile under ubuntu 20.04 with 5.10 kernel

asluchevskiy, 2021-05-18:
### Bug description
Failed to compile under ubuntu 20.04 with 5.10 kernel

### How to reproduce
What are the steps to reproduce the reported issue.
```
git clone https://github.com/snuf/iomemory-vsl4.git
cd iomemory-vsl4
git checkout v5.12.1
make dpkg

...
/home/arceny/iomemory-vsl4/driver_source/kfile.c: In function ‘kfio_create_proc_fops_entry’:
/home/arceny/iomemory-vsl4/driver_source/kfile.c:225:47: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  225 |     return proc_create_data(name, mode, base, (struct file_operations *)fops, data);
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |
      |                                               struct file_operations *
In file included from /home/arceny/iomemory-vsl4/driver_source/kfile.c:37:
./include/linux/proc_fs.h:104:31: note: expected ‘const struct proc_ops *’ but argument is of type ‘struct file_operations *’
  104 | extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
      |                               ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:279: /home/arceny/iomemory-vsl4/driver_source/kfile.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/home/arceny/iomemory-vsl4/driver_source/kcpu.c: In function ‘__kfio_bind_task_to_cpumask’:
/home/arceny/iomemory-vsl4/driver_source/kcpu.c:131:10: error: ‘struct task_struct’ has no member named ‘cpus_allowed’; did you mean ‘nr_cpus_allowed’?
  131 |     tsk->cpus_allowed = *mask;
      |          ^~~~~~~~~~~~
      |          nr_cpus_allowed
/home/arceny/iomemory-vsl4/driver_source/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/home/arceny/iomemory-vsl4/driver_source/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
  214 |     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
      |     ^~~~~~~~
/home/arceny/iomemory-vsl4/driver_source/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
make[4]: *** [scripts/Makefile.build:279: /home/arceny/iomemory-vsl4/driver_source/kcpu.o] Error 1
/home/arceny/iomemory-vsl4/driver_source/kblock.c: In function ‘linux_bdev_update_stats’:
/home/arceny/iomemory-vsl4/driver_source/kblock.c:959:9: error: implicit declaration of function ‘part_stat_lock’ [-Werror=implicit-function-declaration]
  959 |         part_stat_lock();
      |         ^~~~~~~~~~~~~~
/home/arceny/iomemory-vsl4/driver_source/kblock.c:960:9: error: implicit declaration of function ‘part_stat_inc’ [-Werror=implicit-function-declaration]
  960 |         part_stat_inc(GD_PART0, ios[1]);
      |         ^~~~~~~~~~~~~
/home/arceny/iomemory-vsl4/driver_source/kblock.c:960:33: error: ‘ios’ undeclared (first use in this function)
  960 |         part_stat_inc(GD_PART0, ios[1]);
      |                                 ^~~
/home/arceny/iomemory-vsl4/driver_source/kblock.c:960:33: note: each undeclared identifier is reported only once for each function it appears in
/home/arceny/iomemory-vsl4/driver_source/kblock.c:961:9: error: implicit declaration of function ‘part_stat_add’ [-Werror=implicit-function-declaration]
  961 |         part_stat_add(GD_PART0, sectors[1], totalsize >> 9);
      |         ^~~~~~~~~~~~~
/home/arceny/iomemory-vsl4/driver_source/kblock.c:961:33: error: ‘sectors’ undeclared (first use in this function); did you mean ‘sector_t’?
  961 |         part_stat_add(GD_PART0, sectors[1], totalsize >> 9);
      |                                 ^~~~~~~
      |                                 sector_t
/home/arceny/iomemory-vsl4/driver_source/kblock.c:962:33: error: ‘nsecs’ undeclared (first use in this function)
  962 |         part_stat_add(GD_PART0, nsecs[1], kfio_div64_64(duration * HZ, FIO_USEC_PER_SEC));
      |                                 ^~~~~
/home/arceny/iomemory-vsl4/driver_source/kblock.c:963:9: error: implicit declaration of function ‘part_stat_unlock’ [-Werror=implicit-function-declaration]
  963 |         part_stat_unlock();
      |         ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:279: /home/arceny/iomemory-vsl4/driver_source/kblock.o] Error 1
make[3]: *** [Makefile:1807: /home/arceny/iomemory-vsl4/driver_source] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-5.10.16-051016-generic'
make[2]: *** [Makefile:134: modules] Error 2
make[2]: Leaving directory '/home/arceny/iomemory-vsl4/driver_source'
ERROR:
make[1]: *** [debian/rules:102: build-arch-stamp] Error 1
make[1]: Leaving directory '/home/arceny/iomemory-vsl4'
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
make: *** [Makefile:14: dpkg] Error 2

```

### Environment information
Information about the system the module is used on
1. Linux chiahome 5.10.16-051016-generic #202102131334 SMP Sat Feb 13 14:19:45 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
2. gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
3. Ubuntu 20.04.2 LTS (Focal Fossa)
4. v5.12.1

snuf, 2021-05-18:
@asluchevskiy thanks for filing a bug report. There seems to be an issue with your headers or the configuration of the scripts in relation to your headers. Please refer to https://github.com/snuf/iomemory-vsl4/issues/29 and let me know if that resolves the issue.

asluchevskiy, 2021-05-18:
Hi, I checked the `module.lds` file, there was only `module.lds.S.` in my header dir. I made a copy, but the error still exits.

```
arceny@chiahome:~/iomemory-vsl4$ uname -r
5.10.16-051016-generic
arceny@chiahome:~/iomemory-vsl4$ ls -la /usr/src/linux-headers-`uname -r`/scripts/ | grep module.lds
lrwxrwxrwx 1 root root    55 May 18 18:49 module.lds -> ../../linux-headers-5.10.16-051016/scripts/module.lds.S
lrwxrwxrwx 1 root root    55 Feb 13 13:34 module.lds.S -> ../../linux-headers-5.10.16-051016/scripts/module.lds.S
```

snuf, 2021-05-18:
I don't think you can just make a copy because the file is broken then, and will break building, same issue here: https://github.com/snuf/iomemory-vsl/issues/89

asluchevskiy, 2021-05-18:
> I don't think you can just make a copy because the file is broken then, and will break building, same issue here: [snuf/iomemory-vsl#89](https://github.com/snuf/iomemory-vsl/issues/89)

Yes, it helps for me! I removed in the file `module.lds` the last line:

```
#include <asm/module.lds.h>
```

And successfully build the package.

Thank you so much!

snuf, 2021-05-18:
np! happy it worked out :)

closed 2021-05-18 by snuf