RemixVSL/iomemory-vsl4 #34: SX300-3200 failed to start

asluchevskiy, 2021-05-18:
### Bug description
SX300-3200 failed to start

### How to reproduce
sudo dmesg
```   7.507877] fioinf ioDrive 0000:07:00.0: SMP       fpga Microcode BOOT version 0.8.0 1
[    7.507881] fioinf ioDrive 0000:07:00.0: SMP controller Microcode APP  version 0.44.0 0
[    9.428132] fioinf ioDrive 0000:07:00.0: Product UUID is 7f57d177-4e29-5705-9403-3eb459984417
[    9.428137] fioinf ioDrive 0000:07:00.0: Required PCIE bandwidth 4.000 GBytes per sec
[    9.428138] fioinf ioDrive 0000:07:00.0: Board serial number is 1423G0148
[    9.428139] fioinf ioDrive 0000:07:00.0: Adapter serial number is 1423G0148
[    9.428143] fioinf ioDrive 0000:07:00.0: Default capacity        3200.000 GBytes
[    9.428144] fioinf ioDrive 0000:07:00.0: Default sector size     4096 bytes
[    9.428145] fioinf ioDrive 0000:07:00.0: Rated endurance         11.00 PBytes
[    9.428146] fioinf ioDrive 0000:07:00.0: 85C temp range hardware found
[    9.428147] fioinf ioDrive 0000:07:00.0: Maximum capacity        3200.000 GBytes
[    9.428178] fioinf ioDrive 0000:07:00.0: PCIe Slot reported power limit: 75000mWatts
[    9.428179] fioinf ioDrive 0000:07:00.0: PCIe Adapter power limit: 75000mWatts
[    9.428180] fioinf ioDrive 0000:07:00.0: PCIe Adapter power Throttle point: 74750mWatts
[    9.428190] fioinf ioDrive 0000:07:00.0: Firmware Archive Information:
[    9.428191] fioinf ioDrive 0000:07:00.0:   * Name        = ioMemory Firmware 8.9.9.20200113
[    9.428192] fioinf ioDrive 0000:07:00.0:   * Description = ioMemory Firmware 8.9.9.20200113 Archive
[    9.428192] fioinf ioDrive 0000:07:00.0:   * Version     = 8.9.9.0
[    9.428194] fioinf ioDrive 0000:07:00.0:   * Date        = 01/13/2020  Rev 20200113
[    9.428195] fioinf ioDrive 0000:07:00.0: Firmware version 8.9.9 118194 (0x802409 0x1cdb2)
[    9.428197] fioinf ioDrive 0000:07:00.0: Platform version 41 
[    9.428198] fioinf ioDrive 0000:07:00.0: Firmware VCS version 118194 [0x1cdb2]
[    9.428206] fioinf ioDrive 0000:07:00.0: Firmware VCS uid 0xea8892fee031ff5085ff47b5d3b4717e7dd4b603
[    9.449682] fioinf ioDrive 0000:07:00.0: Powercut flush: Supported and Enabled.
[    9.900929] fioinf ioDrive 0000:07:00.0: Loading microcode image 1.0.9 rev 100764 with flags 0x00010100
[    9.901022] fioinf ioDrive 0000:07:00.0: Microcode loaded (0).
[    9.979180] fioinf ioDrive 0000:07:00.0: Multiple queues enabled
[    9.999113] fioinf ioDrive 0000:07:00.0: PCIe power monitor enabled (master). Limit set to 74.750 watts.
[    9.999115] fioinf ioDrive 0000:07:00.0: Thermal monitoring: Enabled
[    9.999116] fioinf ioDrive 0000:07:00.0: Hardware temperature alarm set for 85C.
[   10.145338] fioinf ioDrive 0000:07:00.0: Starting device ioMemory SX300-3200 0000:07:00.0
[   10.145533] fioinf ioMemory SX300-3200 0000:07:00.0: Unable to map request
[   10.145536] fioinf ioMemory SX300-3200 0000:07:00.0: Request page program 000000008bd216e2 invalid -22
[   11.445929] fioerr ioMemory SX300-3200 0000:07:00.0: 12314 Device is unavailable.
[   11.445950] fioerr ioDrive 0000:07:00.0: 12295 Driver start failed with error -11: Try again
[   11.445953] fioinf ioDrive 0000:07:00.0: Controller failed to start (error -11)
[   11.446102] fioerr ioDrive 0000:07:00.0: 12297 Unable to attach PCI (-11).
```

### Possible solution
Is a solution know, or type any plausible suggestions here, if none leave clear.

### Environment information
Information about the system the module is used on
1. Linux chiahome 5.10.16-051016-generic #202102131334 SMP Sat Feb 13 14:19:45 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
3. Ubuntu 20.04
4. v5.12.1
5. FIO device used, if applicable
SX300-3200
07:00.0 Mass storage controller [0180]: SanDisk ioMemory HHHL [1aed:3002]

snuf, 2021-05-18:
Some more info on what hardware you are running might be nice? My guess here is that it's an iommu issue. Try setting ```GRUB_CMDLINE_LINUX_DEFAULT="amd_iommu=on iommu=pt"``` in `/etc/default/grub` and run `update-grub`.

asluchevskiy, 2021-05-18:
I have x570 motherboard and ryzen 9 5900x.
Yes, this was help for me. Is this information provided somewhere in FAQs of the project?

Thank you so much

snuf, 2021-05-18:
np! We're getting to a point that we're seeing several duplicate issues now, so yeah might be time to do a FAQ :)

closed 2021-05-18 by snuf