RemixVSL/iomemory-vsl4 #40: Issue building on Oracle EL 7 with UEK

mattboston, 2022-04-12:
### Bug description
If we try either a "make rpm" or "make dkms" we're getting errors. I would prefer DKMS, but would settle for either. I've also tried the latest official driver and their steps to build DKMS, but it fails as well. 

### How to reproduce
```
git clone https://github.com/snuf/iomemory-vsl4.git
cd iomemory-vsl4
git checkout <tag or some-branch>
make rpm
or 
make dkms
```

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a): Linux dcokvmh01.frtservices.com 5.4.17-2136.305.5.4.el7uek.x86_64 <span>#</span>2 SMP Mon Mar 28 19:02:05 PDT 2022 x86_64 x86_64 x86_64 GNU/Linux
2. The C compiler version used (gcc --version): gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44.0.3)
3. distribution, and version (cat /etc/os-release)
``` NAME="Oracle Linux Server"
VERSION="7.9"
ID="ol"
ID_LIKE="fedora"
VARIANT="Server"
VARIANT_ID="server"
VERSION_ID="7.9"
PRETTY_NAME="Oracle Linux Server 7.9"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:oracle:linux:7:9:server"
HOME_URL="https://linux.oracle.com/"
BUG_REPORT_URL="https://bugzilla.oracle.com/"
ORACLE_BUGZILLA_PRODUCT="Oracle Linux 7"
ORACLE_BUGZILLA_PRODUCT_VERSION=7.9
ORACLE_SUPPORT_PRODUCT="Oracle Linux"
ORACLE_SUPPORT_PRODUCT_VERSION=7.9
```
4. Tag or Branch of iomemory-vsl4 that is being compiled: main (commit 265a128c4d4838c4d1a52591053c2de7df69a733)
5. FIO device used: SanDisk FusionIO SX350
6. Installed Kernel/UEK/DKMS packages:
```[root@dcokvmh01 ~]# rpm -qa | egrep -i 'kernel|uek' | sort
abrt-addon-kerneloops-2.1.11-60.0.1.el7.x86_64
kernel-3.10.0-1160.59.1.el7.x86_64
kernel-3.10.0-1160.62.1.el7.x86_64
[dkms.log](https://github.com/RemixVSL/iomemory-vsl4/files/8474141/dkms.log)
[rpm.log](https://github.com/RemixVSL/iomemory-vsl4/files/8474142/rpm.log)

kernel-debug-devel-3.10.0-1160.62.1.el7.x86_64
kernel-devel-3.10.0-1160.59.1.el7.x86_64
kernel-devel-3.10.0-1160.62.1.el7.x86_64
kernel-headers-3.10.0-1160.62.1.el7.x86_64
kernel-tools-3.10.0-1160.62.1.el7.x86_64
kernel-tools-libs-3.10.0-1160.62.1.el7.x86_64
kernel-uek-5.4.17-2136.305.5.4.el7uek.x86_64
kernel-uek-devel-5.4.17-2136.305.5.4.el7uek.x86_64
kernel-uek-tools-5.4.17-2136.305.5.4.el7uek.x86_64

[root@dcokvmh01 ~]# rpm -qa dkms
dkms-3.0.3-1.el7.noarch
```

mattboston, 2022-04-12:
[dkms.log](https://github.com/RemixVSL/iomemory-vsl4/files/8474158/dkms.log)
[rpm.log](https://github.com/RemixVSL/iomemory-vsl4/files/8474159/rpm.log)

bplein, 2022-04-12:
Don't check out 5.12. Try with main branch.

mattboston, 2022-04-12:
The original post was using the main branch.  Are there any tricks with Oracle Linux 7 with UEK?  Based on the Sandisk docs, I got all the appropriate rpms installed plus some extra ones that weren't documented.

snuf, 2022-04-14:
@mattboston fix should be in `main` now. I reintroduced the check that was missing, which should get you to something that compiles.

mattboston, 2022-04-14:
Just pulled down changes and now getting new errors
[dkms3.log](https://github.com/RemixVSL/iomemory-vsl4/files/8489260/dkms3.log)
.

snuf, 2022-04-14:
@mattboston there is something odd with your headers. `1649939516.510  KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS=1` in the log indicates your kernel uses proc_ops, however when compiling it barfs that they should be file_operations. I tested the code on 5.4.0 (Ubuntu) which uses proc_ops. Can you get the logs from the kfio_config dir for that test?

snuf, 2022-04-14:
I've spun up a test Oracle 7 box. The issue is indeed that it's falsely marking the test as successful, while it should mark it failed:
```[vagrant@fio2 iomemory-vsl4-4.3.7]$ cat kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.log 
[-snip-]
/home/vagrant/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c: In function 'kfioc_has_proc_create_data':
/home/vagrant/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:9:5: warning: passing argument 4 of 'proc_create_data' from incompatible pointer type [enabled by default]
     return proc_create_data(NULL, 0, NULL, pops, NULL);
     ^
In file included from /home/vagrant/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:4:0:
./include/linux/proc_fs.h:45:31: note: expected 'const struct file_operations *' but argument is of type 'const struct proc_ops *'
 extern struct proc_dir_entry *proc_create_data(const char *, umode_t,
                               ^
/home/vagrant/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio_config/KFIOC_X_PROC_CREATE_DATA_WANTS_PROC_OPS/kfioc_test.c:9:5: warning: 'pops' is used uninitialized in this function [-Wuninitialized]
     return proc_create_data(NULL, 0, NULL, pops, NULL);
     ^
[-snip-]
```
It should -Werror there and bomb, but it doesn't.

snuf, 2022-04-14:
@mattboston fix is in `main` now:
```[   71.809152] fioinf ioDrive driver 5.4.17-9379af0-4.3.7.1205     loading...
...
[   74.723843] fioinf ioMemory SX300-1300 0000:00:04.0: probed fct0
[   74.850297] fioinf ioMemory SX300-1300 0000:00:04.0: sector_size=4096
[   74.851137] fioinf ioMemory SX300-1300 0000:00:04.0: Setting channel range data to [2 .. 1023]
[   75.498794] fioinf ioMemory SX300-1300 0000:00:04.0: Found metadata in EBs 348-166, loading...
[   76.411302] fioinf ioMemory SX300-1300 0000:00:04.0: Device has no Virtual Partitions.  Creating compatibility Virtual Partition fioa.
[   76.412482] fioinf ioMemory SX300-1300 0000:00:04.0: Created device of size 1249999998976 bytes with 305175781 sectors of 4096 bytes (180976729 mapped).
[   76.415833] fioinf ioMemory SX300-1300 0000:00:04.0: Creating block device fioa: major: 250 minor: 0 sector size: 4096...
[   76.418950]  fioa: fioa1
[   76.419498] fioinf ioMemory SX300-1300 0000:00:04.0: Exposing Virtual Partition fioa of size 1249999998976 bytes with 305175781 sectors of 4096 bytes (180976729 mapped).
[   76.421310] fioinf ioMemory SX300-1300 0000:00:04.0: Attach succeeded.
```

mattboston, 2022-04-14:
Excellent.  Thank you very much.  Feel free to mark as closed.

mattboston, 2022-04-14:
When I move the cards, I'll re-open or create a new if there are issues.

snuf, 2022-04-14:
Cool!

closed 2022-04-14 by snuf