RemixVSL/iomemory-vsl4 #44: Unable to compile on RHEL8.4 using v4.20.1

TheWitness, 2022-05-31:
### Bug description

Getting a slew of errors with v4.20.1.  Not sure what to do to clear them.  The iomemory-vsl v4.20.3 builds fine, but I don't think I can use that for my system.  Show you the errors below.

### How to reproduce
What are the steps to reproduce the reported issue.
```
git clone https://github.com/snuf/iomemory-vsl4.git
cd iomemory-vsl4
git checkout v4.20.1
make rpm
<snip>
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kcpu.c: In function '__kfio_bind_task_to_cpumask':
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kcpu.c:139:10: error: 'struct task_struct' has no member named 'cpus_allowed'; did you mean 'nr_cpus_allowed'?
     tsk->cpus_allowed = *mask;
          ^~~~~~~~~~~~
          nr_cpus_allowed
make[3]: *** [scripts/Makefile.build:315: /root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kcpu.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c: In function 'linux_bdev_hide_disk':
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1145:13: error: implicit declaration of function 'blk_stop_queue'; did you mean 'blk_sync_queue'? [-Werror=implicit-function-declaration]
             blk_stop_queue(disk->rq);
             ^~~~~~~~~~~~~~
             blk_sync_queue
In file included from ././include/linux/compiler_types.h:58,
                 from <command-line>:
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c: In function 'linux_bdev_update_stats':
./include/linux/compiler-gcc.h:16:19: error: expected expression before '__asm__'
 #define barrier() __asm__ __volatile__("": : :"memory")
                   ^~~~~~~
./include/linux/preempt.h:242:29: note: in expansion of macro 'barrier'
 #define preempt_disable()   barrier()
                             ^~~~~~~
./include/linux/genhd.h:321:26: note: in expansion of macro 'preempt_disable'
 #define part_stat_lock() preempt_disable()
                          ^~~~~~~~~~~~~~~
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1319:19: note: in expansion of macro 'part_stat_lock'
             cpu = part_stat_lock();
                   ^~~~~~~~~~~~~~
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1320:50: error: macro "part_stat_inc" passed 3 arguments, but takes just 2
             part_stat_inc(cpu, &gd->part0, ios[1]);
                                                  ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1320:13: error: 'part_stat_inc' undeclared (first use in this function); did you mean 'param_set_int'?
             part_stat_inc(cpu, &gd->part0, ios[1]);
             ^~~~~~~~~~~~~
             param_set_int
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1320:13: note: each undeclared identifier is reported only once for each function it appears in
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1321:70: error: macro "part_stat_add" passed 4 arguments, but takes just 3
             part_stat_add(cpu, &gd->part0, sectors[1], totalsize >> 9);
                                                                      ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1321:13: error: 'part_stat_add' undeclared (first use in this function); did you mean 'page_ref_add'?
             part_stat_add(cpu, &gd->part0, sectors[1], totalsize >> 9);
             ^~~~~~~~~~~~~
             page_ref_add
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1323:84: error: macro "part_stat_add" passed 4 arguments, but takes just 3
             part_stat_add(cpu, &gd->part0, nsecs[1],   duration * FIO_NSEC_PER_USEC);   // Convert our usec duration to nsecs.
                                                                                    ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1309:29: warning: unused variable 'gd' [-Wunused-variable]
             struct gendisk *gd = disk->gd;
                             ^~
In file included from ././include/linux/compiler_types.h:58,
                 from <command-line>:
./include/linux/compiler-gcc.h:16:19: error: expected expression before '__asm__'
 #define barrier() __asm__ __volatile__("": : :"memory")
                   ^~~~~~~
./include/linux/preempt.h:242:29: note: in expansion of macro 'barrier'
 #define preempt_disable()   barrier()
                             ^~~~~~~
./include/linux/genhd.h:321:26: note: in expansion of macro 'preempt_disable'
 #define part_stat_lock() preempt_disable()
                          ^~~~~~~~~~~~~~~
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1360:19: note: in expansion of macro 'part_stat_lock'
             cpu = part_stat_lock();
                   ^~~~~~~~~~~~~~
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1361:50: error: macro "part_stat_inc" passed 3 arguments, but takes just 2
             part_stat_inc(cpu, &gd->part0, ios[0]);
                                                  ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1362:70: error: macro "part_stat_add" passed 4 arguments, but takes just 3
             part_stat_add(cpu, &gd->part0, sectors[0], totalsize >> 9);
                                                                      ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1364:84: error: macro "part_stat_add" passed 4 arguments, but takes just 3
             part_stat_add(cpu, &gd->part0, nsecs[0],   duration * FIO_NSEC_PER_USEC);
                                                                                    ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1352:29: warning: unused variable 'gd' [-Wunused-variable]
             struct gendisk *gd = disk->gd;
                             ^~
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c: In function 'linux_bdev_update_inflight':
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:1449:18: error: 'struct hd_struct' has no member named 'in_flight'
         gd->part0.in_flight = in_flight;
                  ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c: In function 'kfio_alloc_queue':
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:2323:24: error: incompatible types when assigning to type 'spinlock_t' {aka 'struct spinlock'} from type 'spinlock_t *' {aka 'struct spinlock *'}
         rq->queue_lock = (spinlock_t *)&dp->queue_lock;
                        ^
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c: In function 'linux_bdev_backpressure':
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:2364:16: error: 'struct request_queue' has no member named 'request_fn'; did you mean 'make_request_fn'?
         if (q->request_fn)
                ^~~~~~~~~~
                make_request_fn
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c: In function 'linux_bdev_lock_pending':
/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.c:2418:12: error: 'struct request_queue' has no member named 'request_fn'; did you mean 'make_request_fn'?
     if (q->request_fn == NULL)
            ^~~~~~~~~~
            make_request_fn
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:315: /root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source/kblock.o] Error 1
make[2]: *** [Makefile:1563: _module_/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source] Error 2
make[2]: Leaving directory '/usr/src/kernels/4.18.0-305.el8.x86_64'
make[1]: *** [Makefile:134: modules] Error 2
make[1]: Leaving directory '/root/rpmbuild/BUILD/iomemory-vsl4-4.3.7.1205/driver_source'
+ set +x
ERROR:

error: Bad exit status from /var/tmp/rpm-tmp.n7NsMl (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.n7NsMl (%build)
make: *** [Makefile:20: rpm] Error 1
```

### Possible solution

I've seen some "similar" errors logged in the past.  I have been able to get master to install, but the system then kernel panics.  I've been warned against using the v5.x on a v4.x kernel though, so not surprised.

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a)
    * 4.18.0-305.el8.x86_64
2. The C compiler version used (gcc --version)
    * 8.4.1-1
3. distribution, and version (cat /etc/os-release)
    * 8.4 (Ootpa)
5. Tag or Branch of iomemory-vsl4 that is being compiled
    * v4.20.1
6. FIO device used, if applicable
    * 87:00.0 Mass storage controller [0180]: SanDisk ioMemory Mezzanine [1aed:3003]

TheWitness, 2022-05-31:
Okay, let me try with that.  Like I said, the last time resulted in a kernel panic when I did a `make dkms`.  I'll try the RPM first.

TheWitness, 2022-05-31:
Getting this now.

```sh
Obsoletes: fio-driver iodrive-driver
Processing files: iomemory-vsl4-config-4.18.0-305.el8.x86_64-4.3.7.1205-1.el8.x86_64
Provides: iomemory-vsl4-config iomemory-vsl4-config-4.18.0-305.el8.x86_64 = 4.3.7.1205-1.el8 iomemory-vsl4-config-4.18.0-305.el8.x86_64(x86-64) = 4.3.7.1205-1.el8 iomemory-vsl4-config-4.3.7.1205
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Processing files: iomemory-vsl4-source-4.3.7.1205-1.el8.x86_64
warning: File listed twice: /usr/src/iomemory-vsl4-4.3.7/kfio/x86_64_cc48_libkfio.o_shipped
Provides: config(iomemory-vsl4-source) = 4.3.7.1205-1.el8 iomemory-vsl iomemory-vsl4 iomemory-vsl4-4.3.7.1205 iomemory-vsl4-source = 4.3.7.1205-1.el8 iomemory-vsl4-source(x86-64) = 4.3.7.1205-1.el8 libvsl libvsl.so.1()(64bit)
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Requires: /bin/bash /bin/sh libc.so.6()(64bit) libc.so.6(GLIBC_2.14)(64bit) libc.so.6(GLIBC_2.15)(64bit) libc.so.6(GLIBC_2.2.5)(64bit) libc.so.6(GLIBC_2.3)(64bit) libc.so.6(GLIBC_2.3.4)(64bit) libc.so.6(GLIBC_2.7)(64bit) libgcc_s.so.1()(64bit) libgcc_s.so.1(GCC_3.0)(64bit) libm.so.6()(64bit) libpthread.so.0()(64bit) libpthread.so.0(GLIBC_2.2.5)(64bit) librt.so.1()(64bit) libstdc++.so.6()(64bit) libstdc++.so.6(CXXABI_1.3)(64bit) libstdc++.so.6(GLIBCXX_3.4)(64bit) libstdc++.so.6(GLIBCXX_3.4.11)(64bit) libstdc++.so.6(GLIBCXX_3.4.15)(64bit) libstdc++.so.6(GLIBCXX_3.4.9)(64bit) libuuid.so.1()(64bit) libuuid.so.1(UUID_1.0)(64bit) rtld(GNU_HASH)
Conflicts: iomemory-vsl4
Obsoletes: fio-driver fio-driver-source iodrive-driver iodrive-driver-source
Checking for unpackaged file(s): /usr/lib/rpm/check-files /root/rpmbuild/BUILDROOT/iomemory-vsl4-4.3.7.1205-1.el8.x86_64
error: Installed (but unpackaged) file(s) found:
   /usr/src/iomemory-vsl4-4.3.7/include/kblock_meta.h

RPM build errors:
    File listed twice: /usr/src/iomemory-vsl4-4.3.7/kfio/x86_64_cc48_libkfio.o_shipped
    Installed (but unpackaged) file(s) found:
   /usr/src/iomemory-vsl4-4.3.7/include/kblock_meta.h
make: *** [Makefile:20: rpm] Error 1
```

snuf, 2022-05-31:
Aaaah sorry missed that. kernel panic is not so bueno. The v5 driver, if it compiles, should be fine. Main is the evolution of v5 and v5 is just an evolution and "cleaned up" version of v4.

TheWitness, 2022-05-31:
Understood.  Just posted the make rpm.  Trying `module` now to see if it's just an `rpmbuild` issue.

snuf, 2022-05-31:
it is, the kblock_meta.h is missing from the rpmspec file it seems.

TheWitness, 2022-05-31:
The `make module` worked, but I'm nervous to install it ;)

snuf, 2022-05-31:
you can `insmod` the module and see if it goes belly up. If it does it would be valuable to have the log though, to see what's going on :)

TheWitness, 2022-05-31:
Sure, give me a minute.

TheWitness, 2022-05-31:
Well, it's `insmod /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.ko` is just sitting there hanging.  From /var/log/syslog...

```sh
May 31 08:24:26 rtm-web-01 kernel: [ 5104.323698] fioinf No Queue strategy is set.
May 31 08:24:26 rtm-web-01 kernel: [ 5104.323780] fioinf
May 31 08:24:26 rtm-web-01 kernel: [ 5104.323850] fioinf Copyright (c) 2006-2020 Western Digital Corporation or its affiliates.
May 31 08:24:26 rtm-web-01 kernel: [ 5104.323934] fioinf For Terms and Conditions see the License file included
May 31 08:24:26 rtm-web-01 kernel: [ 5104.324014] fioinf with this driver package.
May 31 08:24:26 rtm-web-01 kernel: [ 5104.324128] fioinf
May 31 08:24:26 rtm-web-01 kernel: [ 5104.324218] fioinf ioDrive driver 4.18.0-def3093-4.3.7.1205     loading...
May 31 08:24:26 rtm-web-01 kernel: [ 5104.324322] fioinf VSL configuration hash: f7771666cf9469344233dd611c787828ea46645e
May 31 08:24:26 rtm-web-01 kernel: [ 5104.325199] fioinf ioDrive 0000:87:00.0: mapping controller on BAR 5
May 31 08:24:26 rtm-web-01 kernel: [ 5104.475595] fioinf ioDrive 0000:87:00.0: Configuring controller.
May 31 08:24:26 rtm-web-01 kernel: [ 5104.475967] fioinf ioDrive 0000:87:00.0: using MSI-X interrupts
May 31 08:24:26 rtm-web-01 kernel: [ 5104.478118] fioinf ioDrive 0000:87:00.0: Starting controller
May 31 08:24:26 rtm-web-01 kernel: [ 5104.526106] fioinf ioDrive 0000:87:00.0: PMP Address: 1 1 1
May 31 08:24:28 rtm-web-01 kernel: [ 5106.542127] fioinf ioDrive 0000:87:00.0: SMP       fpga Microcode BOOT version 0.8.0 1
May 31 08:24:28 rtm-web-01 kernel: [ 5106.542267] fioinf ioDrive 0000:87:00.0: SMP controller Microcode APP  version 0.11.0 0
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694150] fioinf ioDrive 0000:87:00.0: Product UUID is f410c495-19ce-53f8-b34d-0021b2bb8e1e
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694283] fioinf ioDrive 0000:87:00.0: Required PCIE bandwidth 4.000 GBytes per sec
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694410] fioinf ioDrive 0000:87:00.0: Board serial number is 1509D0898
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694525] fioinf ioDrive 0000:87:00.0: Adapter serial number is 1509D0898
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694642] fioinf ioDrive 0000:87:00.0: Default capacity        1250.000 GBytes
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694785] fioinf ioDrive 0000:87:00.0: Default sector size     512 bytes
May 31 08:24:29 rtm-web-01 kernel: [ 5107.694899] fioinf ioDrive 0000:87:00.0: Rated endurance         4.00 PBytes
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695031] fioinf ioDrive 0000:87:00.0: 100C temp range hardware found
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695149] fioinf ioDrive 0000:87:00.0: Maximum capacity        1250.000 GBytes
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695297] fioinf ioDrive 0000:87:00.0: PCIe Slot reported power limit: 0mWatts
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695422] fioinf ioDrive 0000:87:00.0: Warning! PCIe Slot reported power 0mWatts is less than product minimum power 13000mWatts
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695556] fioinf ioDrive 0000:87:00.0: PCIe Adapter power limit: 25000mWatts
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695679] fioinf ioDrive 0000:87:00.0: PCIe Adapter power Throttle point: 24750mWatts
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695810] fioinf ioDrive 0000:87:00.0: Firmware Archive Information:
May 31 08:24:29 rtm-web-01 kernel: [ 5107.695911] fioinf ioDrive 0000:87:00.0:   * Name        = ioMemory Firmware 8.9.9.20180621
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696039] fioinf ioDrive 0000:87:00.0:   * Description = ioMemory Firmware 8.9.9.20180621 Archive
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696182] fioinf ioDrive 0000:87:00.0:   * Version     = 8.9.9.0
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696284] fioinf ioDrive 0000:87:00.0:   * Date        = 06/21/2018  Rev 20180621
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696410] fioinf ioDrive 0000:87:00.0: Firmware version 8.9.9 118194 (0x802409 0x1cdb2)
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696537] fioinf ioDrive 0000:87:00.0: Platform version 44
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696636] fioinf ioDrive 0000:87:00.0: Firmware VCS version 118194 [0x1cdb2]
May 31 08:24:29 rtm-web-01 kernel: [ 5107.696765] fioinf ioDrive 0000:87:00.0: Firmware VCS uid 0xea8892fee031ff5085ff47b5d3b4717e7dd4b603
May 31 08:24:29 rtm-web-01 kernel: [ 5107.710792] fioinf ioDrive 0000:87:00.0: Powercut flush: Supported and Enabled.
May 31 08:24:29 rtm-web-01 kernel: [ 5107.987914] fioinf ioDrive 0000:87:00.0: Loading microcode image 1.0.9 rev 100764 with flags 0x00010100
May 31 08:24:29 rtm-web-01 kernel: [ 5107.988174] fioinf ioDrive 0000:87:00.0: Microcode loaded (0).
May 31 08:24:29 rtm-web-01 kernel: [ 5108.030674] fioinf ioDrive 0000:87:00.0: Limiting to 4 banks
May 31 08:24:29 rtm-web-01 kernel: [ 5108.030760] fioinf ioDrive 0000:87:00.0: Multiple queues enabled
May 31 08:24:29 rtm-web-01 kernel: [ 5108.059179] fioinf ioDrive 0000:87:00.0: PCIe power monitor enabled (master). Limit set to 24.750 watts.
May 31 08:24:29 rtm-web-01 kernel: [ 5108.059292] fioinf ioDrive 0000:87:00.0: Thermal monitoring: Enabled
May 31 08:24:29 rtm-web-01 kernel: [ 5108.059378] fioinf ioDrive 0000:87:00.0: Hardware temperature alarm set for 100C.
May 31 08:24:29 rtm-web-01 kernel: [ 5108.155305] fioinf ioDrive 0000:87:00.0: Starting device HP 1.2TB Read Intensive Mezzanine PCIe Workload Accelerator for BladeSystem c-Class 0000:87:00.0
[... cut ...]
May 31 08:25:51 rtm-web-01 kernel: [ 5189.363957] fioinf HP 1.2TB Read Intensive Mezzanine PCIe Workload Accelerator for BladeSystem c-Class 0000:87:00.0: Device has no Virtual Partitions.  Creating compatibility Virtual Partition fioa.
May 31 08:25:53 rtm-web-01 kernel: [ 5191.750626] fioinf HP 1.2TB Read Intensive Mezzanine PCIe Workload Accelerator for BladeSystem c-Class 0000:87:00.0: Created device of size 1250000000000 bytes with 2441406250 sectors of 512 bytes (1350522315 mapped).
```

So, not kernel panicking, but doing what cause the kernel panic and just timing out in it's attempt to load.

TheWitness, 2022-05-31:
lol, spoke too soon.  Dead.  Get you the information after I hit the light's out button and it comes back up.

TheWitness, 2022-05-31:
It rebooted, but nothing in `/var/log/syslog`.  So, not sure why it's not loading.

TheWitness, 2022-05-31:
Hmm, our RHEL8 is pretty custom.  Let me see if I can find how to set that up.

TheWitness, 2022-05-31:
Okay, setup a kern.* path and restarted rsyslog.  The last thing before the reboot is this.  Nothing special:

`May 31 11:31:35 rtm-web-01 kernel: [10970.934855] fioinf HP 1.2TB Read Intensive Mezzanine PCIe Workload Accelerator for BladeSystem c-Class 0000:87:00.0: Created device of size 1250000000000 bytes with 2441406250 sectors of 512 bytes (1350522315 mapped).`

TheWitness, 2022-05-31:
I was able to `insmod` using this branch https://github.com/RemixVSL/iomemory-vsl4/tree/BLK_ALLOC_QUEU-vsl.  So, something in the last 26 commit's causes RHEL8 to puke.

TheWitness, 2022-06-01:
I've now got dkms rolling along too.

snuf, 2022-06-01:
4.20.3 for VSL is entirely different from 4.20.1 VSL4. VSL4 is just a tooling update for a slightly patched original driver, whereas VSL 4.20.3  contains a lot of changes. The VSL driver's 4.20.3 was actually pulled forward, that said it is also stale now, and has been working from main for Rocky/RHEL/CentOS, which some hiccups here and there.
 
The branch you're referring to includes a lot of changes around the disk creation, which may be the problem. That said it would be helpful if you could do a `make module` from main and post the config output that it shows. It tells me which things are available and are not, which makes tracing the code path a bit easier.

TheWitness, 2022-06-01:
I'll forward along hopefully later today.

TheWitness, 2022-06-01:
Output of `make module` attached.

[make.module.txt](https://github.com/RemixVSL/iomemory-vsl4/files/8818351/make.module.txt)

snuf, 2022-06-01:
That part does not live in the driver as such, but in the binary blob of the driver. The parts that do are related to how the device is exposed, and what kernel function parts are used from the Linux kernel to interact with. The order of those calls, and how they work together changes over time, things get introduced others get removed. RedHat/CentOS/Rocky Linux have a habit of keeping old stuff around that they patch. So they become a mix of old and new things. It could be that a specific combination in your specific kernel triggers this behavior. but someone running Rocky Linux with 4.20 does not encounter the problem.

snuf, 2022-06-03:
I would not bet on it, maybe if they move to 9 they will adopt bio_submit instead of make_request_fn finally.... I'll see when I have time to make a patch to put a change in main.

snuf, 2022-09-08:
For reference the older module from branch `fio-original-4.3.7.1205` compiles fine on older Rhel based kernels (4.18/4.19/4.20).
```
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: could not find /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio/.x86_64_cc48_libkfio.o.cmd for /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio/x86_64_cc48_libkfio.o
  CC      /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.mod.o
  LD [M]  /root/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/iomemory-vsl4.ko
make[1]: Leaving directory `/usr/src/kernels/4.19.0+1-x86_64'
[10:19 xcp-ng-agkjbbcs iomemory-vsl4-4.3.7]# uname -a
Linux xcp-ng-agkjbbcs 4.19.0+1 #1 SMP Thu Jan 13 12:55:45 CET 2022 x86_64 x86_64 x86_64 GNU/Linux
[10:43 xcp-ng-agkjbbcs iomemory-vsl4-4.3.7]# git branch -v
* fio-original-4.3.7.1205 7e12eef Create .gitignore
  main                    fe767c6 Update README.md
[10:43 xcp-ng-agkjbbcs iomemory-vsl4-4.3.7]# 
```

closed 2022-09-08 by snuf