RemixVSL/iomemory-vsl4 #45: No rule to make target 'clean' in /usr/src/linux-5.15.43-Unraid when making module from source

Pihg, 2022-06-11:
Following a mix of directions from: https://wiki.unraid.net/Building_a_custom_kernel (though deprecated, it helped me to get seemingly requisite packages installed. I made it through 'make clean', didn't do 'make menuconfig', and I'm not trying to make a whole new kernel) and this Github page here for VSL4, making from source, I've run into the following issue (see attached image)
[image]

It seems the Makefile that exists for my given kernel headers is missing some information.

I know enough about Linux to be dangerous, but not enough about compiling to sort this one out. 

Apologies if this isn't the right spot, as it's more of an issue with my environment than the iomemory-vsl4 module as a whole. I've gotten it to compile successfully inside an Ubuntu VM, however the resultant module isn't compatible with Unraid as it ends up being from a different kernel-header base. 

I see a split solution: fix this compiling issue in my underlying Unraid build environment, or get Ubuntu to use force-fed headers for 5.15.43. I've tried the latter to no avail, as it still seems to flag/tag/use the running Ubuntu headers to make the module and Unraid refuses to load the module.

snuf, 2022-12-22:
Look at #2 [here](https://wiki.archlinux.org/title/Compile_kernel_module), and you should end up with a set of headers you can use to compile the module. I'm assuming you have the Unraid kernel source in `/usr/src/linux-5.15.43-Unraid` directory there?

Pihg, 2022-12-22:
I haven’t touched that particular project in so long, I’ll have to check into it again and let you know. I’ve had a hard time finding direct source for unRAID in the past, I think they keep their customizations close to the chest which…doesn’t seem accurate.

THANK YOU nonetheless, for the suggestion above. I’ll update here with what I find.

snuf, 2022-12-22:
@Pihg yeah I've looked at it too in the past and at some point they made it "hard" to get the stuff required to do things, which is not great indeed.

snuf, 2023-09-08:
closing due to age.

closed 2023-09-08 by snuf