RemixVSL/iomemory-vsl4 #46: No rule to make target

nerdalertdk, 2022-07-06:
### Bug description

Trying to build dkms, but fails.
```shell
make dkms / fails
make dpkg / fails
make module / works
dkms build -m iomemory-vsl4 -v 5.10.0-15-fe767c6 / works
```

### How to reproduce
What are the steps to reproduce the reported issue.
```shell
git clone https://github.com/snuf/iomemory-vsl4.git
cd iomemory-vsl4
make dkms

```

```shell
Building module:
cleaning build area...
'make' DKMS_KERNEL_VERSION=5.10.0-15-amd64.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-15-amd64 (x86_64)
Consult /var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/make.log for more information.
make[1]: *** [Makefile:121: dkms] Error 10
make[1]: Leaving directory '/opt/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make: *** [Makefile:8: dkms] Error 2
```

Last lines of make.log
```shell
/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kcpu.c: In function ‘kfio_map_cpus_to_read_queues’:
/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_hist’ [-Wvla]
  214 |     uint32_t node_hist[nodes_possible], node_map[nodes_possible];
      |     ^~~~~~~~
/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kcpu.c:214:5: warning: ISO C90 forbids variable length array ‘node_map’ [-Wvla]
make[5]: *** No rule to make target '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kfio/x86_64_cc_libkfio.o', needed by '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/iomemory-vsl4.o'.  Stop.
make[5]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kcsr.o
make[4]: *** [/usr/src/linux-headers-5.10.0-15-common/Makefile:1846: /var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build] Error 2
make[3]: *** [/usr/src/linux-headers-5.10.0-15-common/Makefile:185: __sub-make] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-5.10.0-15-amd64'
make[2]: *** [Makefile:136: modules] Error 2
make[2]: Leaving directory '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build'
```

dkms build
```shell
root@nas1:/opt/iomemory-vsl4# dkms build -m iomemory-vsl4 -v 5.10.0-15-fe767c6

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' DKMS_KERNEL_VERSION=5.10.0-15-amd64.....
cleaning build area...

DKMS: build completed.
```

### Environment information
Information about the system the module is used on
1. Linux kernel compiled against (uname -a): `5.10.0-15-amd64`
2. The C compiler version used (gcc --version) `gcc (Debian 10.2.1-6) 10.2.1 20210110`
3. distribution, and version (cat /etc/os-release) `Debian GNU/Linux 11 (bullseye)`
4. Tag or Branch of iomemory-vsl4 that is being compiled `main´
5. FIO device used, if applicable
   * fio-status
   * lspci -b -nn

nerdalertdk, 2022-07-06:
It seams its my libkfio thats not correct discoverd.

```shell
make[5]: *** No rule to make target '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kfio/x86_64_cc_libkfio.o', needed by '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/iomemory-vsl4.o'.  Stop.
```

```shell
root@nas1:/opt/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7/kfio# ls -l /var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kfio/
total 17732
-rw-r--r-- 1 root root  227608 Jul  6 15:00 function_hashes.txt
-rwxr-xr-x 1 root root    4474 Jul  6 15:00 kfio_operations.sh
-rw-r--r-- 1 root root    1706 Jul  6 15:00 README.md
-rw-r--r-- 1 root root 1592640 Jul  6 15:00 x86_64_cc41_libkfio.o_shipped
-rw-r--r-- 1 root root 1644616 Jul  6 15:00 x86_64_cc43_libkfio.o_shipped
-rw-r--r-- 1 root root 1648912 Jul  6 15:00 x86_64_cc44_libkfio.o_shipped
-rw-r--r-- 1 root root 1563952 Jul  6 15:00 x86_64_cc46_libkfio.o_shipped
-rw-r--r-- 1 root root 1620392 Jul  6 15:00 x86_64_cc48_libkfio.o_shipped
-rw-r--r-- 1 root root 1623328 Jul  6 15:00 x86_64_cc49_libkfio.o_shipped
-rw-r--r-- 1 root root 1641896 Jul  6 15:00 x86_64_cc54_libkfio.o_shipped
-rw-r--r-- 1 root root 1638760 Jul  6 15:00 x86_64_cc63_libkfio.o_shipped
-rw-r--r-- 1 root root 1638760 Jul  6 15:00 x86_64_cc75_libkfio.o_shipped
-rw-r--r-- 1 root root 1638760 Jul  6 15:00 x86_64_latest_libkfio.o_shipped
-rw-r--r-- 1 root root 1638760 Jul  6 15:00 x86_64_libkfio.o_shipped
```

snuf, 2022-07-06:
@nerdalertdk ah yes I was looking into this as someone bumped into this in iomemory-vsl. The problem seems to be the execution of the `module_operations.sh` from the `Makefile`. Something changed where the `./module_operations.sh (yada yada)` breaks and does not receive the environment from the `Makefile` anymore. When using $(shell ./module_operations.sh (yada yada)` it works, but you don't get the output. I'm still looking into this, but it's kind of interesting that there seems to be a change in how environment gets propagated, or rather not.

nerdalertdk, 2022-07-06:
@snuf sounds good :)
is there an ugly quick fix :D or do you need any help debugging ?

snuf, 2022-07-06:
@nerdalertdk quick dirty fix is to replace: https://github.com/RemixVSL/iomemory-vsl4/blob/fe767c61da24478a048d2d6255357c96cc123ed7/root/usr/src/iomemory-vsl4-4.3.7/Makefile#L121
with:
```$(shell ./module_operations.sh -n "$(FIO_DRIVER_NAME)" -v "$(MODULE_VERSION)" -d)```
although it suppresses the output of the `module_operations` so need to find a nicer way, if you know of one I'm all ears :)

nerdalertdk, 2022-07-06:
Now i'm getting this error, which i don't understand 

```shell
 not 5.10.0-15-fe767c6-4.3.7 Warning:   patched with  not 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Ok:  already patched with 5.10.0-15-fe767c6-4.3.7 Adding, buidling and installing iomemory-vsl4/5.10.0-15-fe767c6 with DKMS  Kernel preparation unnecessary for this kernel.  Skipping...  Building module: cleaning build area... 'make' DKMS_KERNEL_VERSION=5.10.0-15-amd64.....(bad exit status: 2)
/bin/sh: 1: Syntax error: "(" unexpected
make[1]: *** [Makefile:121: dkms] Error 2
make[1]: Leaving directory '/opt/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
make: *** [Makefile:8: dkms] Error 2
root@nas1:/opt/iomemory-vsl4#
```

```shell
.PHONY: add_module_version patch_module_version
add_module_version patch_module_version:
	@ret="$$("$(shell pwd)"/module_operations.sh -n "$(FIO_DRIVER_NAME)" -v "$(MODULE_VERSION)")" ; \
	if [ -n "$$ret" ] ; then \
		echo "" ; \
		echo ""$$ret"" ; \
		echo "" ; \
	fi

.PHONY: dkms
dkms:
	$(shell ./module_operations.sh -n "$(FIO_DRIVER_NAME)" -v "$(MODULE_VERSION)" -d)

.PHONY: dpkg
dpkg:
	# patch fio_version, fio_short_version in debian/fio_values
	cd $(shell git rev-parse --show-toplevel) && \
			$(MAKE) dpkg

```

nerdalertdk, 2022-07-06:
it's the same error in the log file

snuf, 2022-07-06:
o_O

nerdalertdk, 2022-07-06:
It did fill out the module_operations in my original log 

```shell
make[1]: Entering directory '/opt/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7'
./module_operations.sh -n "iomemory-vsl4" -v "fe767c6" -d
```

nerdalertdk, 2022-07-06:
Doing just `root@nas1:/opt/iomemory-vsl4/root/usr/src/iomemory-vsl4-4.3.7# ./module_operations.sh -n "iomemory-vsl4" -v "fe767c6" -d` gives same error 

`make[5]: *** No rule to make target '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/kfio/x86_64_cc_libkfio.o', needed by '/var/lib/dkms/iomemory-vsl4/5.10.0-15-fe767c6/build/iomemory-vsl4.o'.  Stop.`

nerdalertdk, 2022-07-07:
Think it might be my CROSS_COMPILE thats not set

```shell
ifneq ($(CROSS_COMPILE),)
FIOCC=$(CROSS_COMPILE)gcc
else
FIOCC=$(CC)
endif
```

```shell
root@nas1:/usr/src/iomemory-vsl4-5.10.0-15-fe767c6# printenv
SHELL=/bin/bash
LANGUAGE=en_US:en
MAKE_TERMOUT=/dev/pts/1
PWD=/usr/src/iomemory-vsl4-5.10.0-15-fe767c6
LOGNAME=root
HOME=/root
LANG=en_US.UTF-8
MFLAGS=-w
MAKEFLAGS=w
TERM=xterm-256color
USER=root
MAKE_TERMERR=/dev/pts/1
SHLVL=2
MAKELEVEL=2
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
MAIL=/var/mail/root
OLDPWD=/opt/iomemory-vsl4
_=/usr/bin/printenv
```

nerdalertdk, 2022-07-07:
So fixed it.

This was the fix

`make CC=gcc dkms`

closed 2022-07-07 by nerdalertdk