RemixVSL/iomemory-vsl4 #47: 60-persistent-fio.rules doesnt work on proxmox

kino0924, 2022-08-25:
### Bug description
I am trying to add 60-persistent-fio.rules into udev.
https://github.com/RemixVSL/iomemory-vsl/issues/74

### How to reproduce
What are the steps to reproduce the reported issue.
```
git clone https://github.com/snuf/iomemory-vsl4.git
cd iomemory-vsl4
make dkms

"reboot"

cp iomemory-vsl4/tools/udev/rules.d/60-persistent-fio.rules /etc/udev/rules.d/

"reboot"

takes forever to boot
and syslog contains this

Aug 25 03:31:32 pve systemd-udevd[574]: fioa1: Worker [588] processing SEQNUM=4954 is taking a long time
Aug 25 03:32:40 pve systemd-udevd[574]: fioa1: Worker [588] processing SEQNUM=4954 killed
Aug 25 03:32:40 pve systemd-udevd[574]: Worker [588] terminated by signal 9 (KILL)
Aug 25 03:32:40 pve systemd-udevd[574]: fioa1: Worker [588] failed

also, despite of persistant-fio.rules, I see this error on dmesg

[   11.889181] ================================================================================
[   11.889244] UBSAN: array-index-out-of-bounds in /var/lib/dkms/iomemory-vsl4/5.15.39-4-fe767c6/build/kcpu.c:257:29
[   11.889274] index 1 is out of range for type 'uint32_t [*]'
[   11.889292] CPU: 0 PID: 319 Comm: kworker/0:5 Tainted: P        W IO      5.15.39-4-pve #1
[   11.889294] Hardware name: Dell Inc. Precision 5820 Tower/002KVM, BIOS 1.11.1 04/24/2019
[   11.889296] Workqueue: events work_for_cpu_fn
[   11.889306] Call Trace:
[   11.889308]  <TASK>
[   11.889311]  dump_stack_lvl+0x4a/0x63
[   11.889320]  dump_stack+0x10/0x16
[   11.889322]  ubsan_epilogue+0x9/0x49
[   11.889324]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   11.889327]  ? __alloc_pages+0x17b/0x320
[   11.889332]  kfio_map_cpus_to_read_queues+0x1b2/0x630 [iomemory_vsl4]
[   11.889386]  ? dma_direct_alloc+0xee/0x2c0
[   11.889391]  ifio_16f86.0f44428653ea0b4766f7b44db386e2dd47d.4.3.7.1205+0xb8/0xd0 [iomemory_vsl4]
[   11.889427]  ifio_25005.a659bcb9c7a6b0c8868f9462e81739ff7b7.4.3.7.1205+0xff/0x6e0 [iomemory_vsl4]
[   11.889460]  ifio_75996.b4e60ba8ffd03bc396fde18541f7e3f09f3.4.3.7.1205+0xa37/0x1550 [iomemory_vsl4]
[   11.889489]  ? kfio_vmalloc+0xe/0x20 [iomemory_vsl4]
[   11.889510]  ? vmalloc+0x21/0x30
[   11.889516]  ifio_d8156.024cd992cd73ed3fc5751d93a4015c5a808.4.3.7.1205+0x5f8/0x730 [iomemory_vsl4]
[   11.889544]  ? acpi_register_gsi_ioapic+0x94/0x180
[   11.889550]  ? __request_region+0x68/0xa0
[   11.889554]  ? __pci_request_region+0x17a/0x340
[   11.889559]  iodrive_pci_attach+0x2a/0x330 [iomemory_vsl4]
[   11.889586]  ? __pci_request_selected_regions+0x40/0x80
[   11.889590]  iodrive_pci_probe+0x70/0x160 [iomemory_vsl4]
[   11.889608]  local_pci_probe+0x48/0x90
[   11.889611]  work_for_cpu_fn+0x17/0x30
[   11.889615]  process_one_work+0x228/0x3d0
[   11.889617]  worker_thread+0x223/0x420
[   11.889619]  ? process_one_work+0x3d0/0x3d0
[   11.889621]  kthread+0x127/0x150
[   11.889624]  ? set_kthread_struct+0x50/0x50
[   11.889628]  ret_from_fork+0x1f/0x30
[   11.889633]  </TASK>
[   11.889634] ================================================================================

** poof, broken token **
```

### Environment information
Information about the system the module is used on
[... 134 lines cut ...]
0000:64:0c.5 System peripheral [0880]: Intel Corporation Sky Lake-E LM Channel 1 [8086:2045] (rev 04)
0000:64:0c.6 System peripheral [0880]: Intel Corporation Sky Lake-E LMS Channel 1 [8086:2046] (rev 04)
0000:64:0c.7 System peripheral [0880]: Intel Corporation Sky Lake-E LMDP Channel 1 [8086:2047] (rev 04)
0000:64:0d.0 System peripheral [0880]: Intel Corporation Sky Lake-E DECS Channel 2 [8086:2048] (rev 04)
0000:64:0d.1 System peripheral [0880]: Intel Corporation Sky Lake-E LM Channel 2 [8086:2049] (rev 04)
0000:64:0d.2 System peripheral [0880]: Intel Corporation Sky Lake-E LMS Channel 2 [8086:204a] (rev 04)
0000:64:0d.3 System peripheral [0880]: Intel Corporation Sky Lake-E LMDP Channel 2 [8086:204b] (rev 04)
0000:65:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM204 [GeForce GTX 980] [10de:13c0] (rev a1)
0000:65:00.1 Audio device [0403]: NVIDIA Corporation GM204 High Definition Audio Controller [10de:0fbb] (rev a1)
0000:b2:00.0 PCI bridge [0604]: Intel Corporation Sky Lake-E PCI Express Root Port A [8086:2030] (rev 04)
0000:b2:05.0 System peripheral [0880]: Intel Corporation Sky Lake-E VT-d [8086:2034] (rev 04)
0000:b2:05.2 System peripheral [0880]: Intel Corporation Sky Lake-E RAS Configuration Registers [8086:2035] (rev 04)
0000:b2:05.4 PIC [0800]: Intel Corporation Sky Lake-E IOxAPIC Configuration Registers [8086:2036] (rev 04)
0000:b2:05.5 RAID bus controller [0104]: Intel Corporation Volume Management Device NVMe RAID Controller [8086:201d] (rev 04)
0000:b2:12.0 Performance counters [1101]: Intel Corporation Sky Lake-E M3KTI Registers [8086:204c] (rev 04)
0000:b2:12.1 Performance counters [1101]: Intel Corporation Sky Lake-E M3KTI Registers [8086:204d] (rev 04)
0000:b2:12.2 System peripheral [0880]: Intel Corporation Sky Lake-E M3KTI Registers [8086:204e] (rev 04)
0000:b2:15.0 System peripheral [0880]: Intel Corporation Sky Lake-E M2PCI Registers [8086:2018] (rev 04)
0000:b2:16.0 System peripheral [0880]: Intel Corporation Sky Lake-E M2PCI Registers [8086:2018] (rev 04)
0000:b2:16.4 System peripheral [0880]: Intel Corporation Sky Lake-E M2PCI Registers [8086:2018] (rev 04)
0000:b2:17.0 System peripheral [0880]: Intel Corporation Sky Lake-E M2PCI Registers [8086:2018] (rev 04)
0000:b3:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Oland [Radeon HD 8570 / R5 430 OEM / R7 240/340 / Radeon 520 OEM] [1002:6611]
0000:b3:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Oland/Hainan/Cape Verde/Pitcairn HDMI Audio [Radeon HD 7000 Series] [1002:aab0]
10000:00:02.0 PCI bridge [0604]: Intel Corporation Sky Lake-E PCI Express Root Port C [8086:2032] (rev 04)
10000:00:03.0 PCI bridge [0604]: Intel Corporation Sky Lake-E PCI Express Root Port D [8086:2033] (rev 04)

snuf, 2022-12-22:
I've not seen this before. USBAN points at something going wrong with the thread allocation. The main question is if USBAN takes place before udev hangs itself and shoots the `fioa1` Worker? that said that piece of code USBAN stumbles on has changed on `main`, might be worth pulling latest main and checking if that fixes it?

snuf, 2023-01-24:
fix for thread issue is in main.

closed 2023-01-24 by snuf